MCUdude/MiniCore #105: Cannot compile arduino-lmic

Donderda, 2019-08-02:
**Arduino IDE: ** 1.8.9
**MiniCore Version: ** 2.0.3 
**Processor: ** ATmega328PB @8MHz external; Running at 3.3V.

Hi everyone,

I have a problem compiling this library with MiniCore: https://github.com/matthijskooijman/arduino-lmic

When I try to compile the ttn-abp example, I get these errors:

```
Build options changed, rebuilding all
libraries/arduino-lmic/lmic/lmic.c.o: In function `calcRxWindow':
/Users/me/Documents/Arduino/libraries/arduino-lmic/src/lmic/lmic.c:2324: undefined reference to `table_get_ostime'
/Users/me/Documents/Arduino/libraries/arduino-lmic/src/lmic/lmic.c:2324: undefined reference to `table_get_ostime'
libraries/arduino-lmic/lmic/lmic.c.o: In function `calcBcnRxWindowFromMillis':
/Users/me/Documents/Arduino/libraries/arduino-lmic/src/lmic/lmic.c:2324: undefined reference to `table_get_ostime'
libraries/arduino-lmic/lmic/lmic.c.o: In function `schedRx12':
/Users/me/Documents/Arduino/libraries/arduino-lmic/src/lmic/lmic.c:2324: undefined reference to `table_get_ostime'
libraries/arduino-lmic/lmic/lmic.c.o: In function `initDefaultChannels':
/Users/me/Documents/Arduino/libraries/arduino-lmic/src/lmic/lmic.c:2324: undefined reference to `table_get_u4'
libraries/arduino-lmic/lmic/radio.c.o: In function `radio_irq_handler':
/Users/me/Documents/Arduino/libraries/arduino-lmic/src/lmic/radio.c:777: undefined reference to `table_get_u2'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATmega328.
```

Here are my settings:
[image]

When I try to compile it for my Uno, everything works as expected. Do you have any idea why this is happening? 

Or do I have to contact the author of this library?

Thank you for your time.

p.s.: I also doesn't compile when I change the chip variant

Donderda, 2019-08-02:
p.p.s.: I also tried to compile it with enabled / disabled LTO settings.

Donderda, 2019-08-02:
I was able to fix this by using platformio rather than the Arduino IDE.

MCUdude, 2019-08-02:
Well, I'll still have to investigate what's wrong. I'll keep the issue open until I know what it is.

MCUdude, 2019-11-16:
It seems to be a library related issue. I get the same error when trying to compile for Arduino Uno.
Related issue: https://github.com/matthijskooijman/arduino-lmic/issues/243.

Conclusion: noting wrong with MiniCore.

closed 2019-11-16 by MCUdude