MCUdude/MiniCore #106: MiniCore, Have I damaged my chip or bricked i? (newbie)

andymouse123, 2019-08-27:
Hi All
I have been using MiniCore quite successfully, I find it easy to use and great but my last Two MCU's will now not burn bootloader, upload sketches and wont respond to Nick gammons great Board Detector Program, yet the software that did successfully load is still running ok, I am using a "bare bones " set up an 328p_AU (on a breakout board" with decoupling caps and a 10k pull up on reset. I have two other boards set at 1 Mhz internal and they are working at the moment but would like to no what I have done because I don't want to repeat it, do you think the un responsive board has had it? I have also tried a FTDI USB to serial board and that doesn't work either. Is it possible I have "Bricked" the chip? I am quite a newbie and running out of ideas. if it helps I will show you the code I am using, it simply has two switches to switch between deep sleep and awake with an LED my goal is to run on batteries  so the deep sleep is what i'm playing with but afraid that I'm doing something wrong and can I fix the chip that runs but wont talk to me?..Please help as I was really getting along nicely and learning a lot
Andy

[code]

/*
  DDRD |= 1 << 5; is an example of setting a bit in the "data direction register" in this case port D bit 5 using an "OR" all other bits unchanged
  DDRD &= ~(1 <<5); is an example of clearing a bit in the "data direction register" in this case port D bit 5 using an " AND NOT" all other bits unchanged

  Whole object here is to use port manipulation instead of "arduino functions" (as much as I no how to at moment)

  Andy Browns first go at direct port manipulation so dont take the micky!
*/

#include "avr/sleep.h"
#define wake_pin 3                 // AU chip, pin 1 (PD3)
#define sleep_pin 4                // AU chip, pin 2 (PD4)
#define led_pin 5                  // AU chip, pin 9 (PD5)

void setup()
{
  DDRD = 0x20;                     // setting all pins of port D to 0 (INPUTS) exept the led pin which is set to 1 (OUTPUT)
  PORTD = 0x38;                    // setting wake_pin, sleep_pin, led_pin to 1 (HIGH)
}

void loop()
{
  if (bit_is_clear(PIND, sleep_pin))                 // checks to see if sleep_pin button has been pressed setting sleep_low
  {
    goto_sleep();                                    // call sleep function
  }
  PORTD |= 1 << led_pin;                             // sets led_pin to 1 (HIGH)
}

void goto_sleep()                                    // sleep function
{
  PORTD &= ~(1 << led_pin);                          // sets led_pin to 0 (LOW)
  ADCSRA &= ~(1 << 7);                               // by writing bit 7 to zero, the ADC is turned off
  set_sleep_mode (SLEEP_MODE_PWR_DOWN);              // AVR power mode
  cli();                                             // clear interupts before attaching interupts
  sleep_enable();
  attachInterrupt(digitalPinToInterrupt(wake_pin), sleep_wakeup, LOW);         // attach interupt to wake pin and check for low on wake_pin if true then goto ISR
  MCUCR = bit (BODS) | bit (BODSE);                  // this stuff swithes off "brown out" detection
  MCUCR = bit (BODS);                                // must be done within 4cycles of the above
  sei();                                             // sets interupts
  sleep_cpu();                                       // cpu stops here until interupt sensed, then sleep_wakeup (ISR)
}

void sleep_wakeup()                                  // ISR ( interupt service routine) function
{
  sleep_disable();
  detachInterrupt(digitalPinToInterrupt(wake_pin));  // disingage interupts
  ADCSRA |= (1 << 7);                                // by writing bit 7 to one, the ADC is turned on
}
[/code]

MCUdude, 2019-08-27:
So you're not able to re-burn the bootloader even if you physically connect PD4 to VCC, so that `goto_sleet()` is never invoked?

andymouse123, 2019-08-27:
I have not tried that ..thank you, I shall in the morning and let you now how I get on, that is an interesting thought .

andymouse123, 2019-08-28:
Hi 
I have tried you suggestion and no different I'm afraid, the chip is still running the software above perfectly but will not respond to a burn bootloader, it wont respond to Nick Gamons board detector, and won't accept a new sketch but runs fine,is it possible for an accidental reverse polarity of the power supply to cause it to not respond but still run software?..Im stumped any thoughts I could send the error messages to you?

Andy

andymouse123, 2019-08-29:
Hi,
I put a good chip on a breadboard and firstly set up Nick Gammon's board detector routine and all was well, I then changed nothing  and placed the suspect board (which is still running sketch fine) into the breadboard and got the "could not enter programming mode" so then I tried to upload a sketch using Arduino as an ISP and got the following error message

`Sketch uses 830 bytes (2%) of program storage space. Maximum is 32256 bytes.
Global variables use 13 bytes (0%) of dynamic memory, leaving 2035 bytes for local variables. Maximum is 2048 bytes.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -carduino -PCOM5 -b19200 -Uflash:w:C:\Users\andy\AppData\Local\Temp\arduino_build_586113/sleep_test3.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

An error occurred while uploading the sketch
Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.`

I am baffled,  but now will solder on another chip to a breakout board and try that again, I suspect it will be fine but would love to no what could have caused this

Regards

Andy

andymouse123, 2019-09-01:
thanks to all for your help ! I think used the prescaler in correctly

andymouse123, 2019-09-02:
Hi
I was a bit premature and got mixed up I still have an issue but I have verified the wiring.
Still no solution i'm afraid,but what I did do is solder a new 328p-AU onto a new breakout board, I then set an UNO as an ISP as normal and wired up the chip appropriately and the chip responded no problem so I went on an used Arduino as an ISP and uploaded a sketch to the new chip and again no problem. So I no that there is no wiring error because I simply replaced new chip with the troublesome one and it was unable to enter program mode. At this point I dont no what to do and i don't simply put in the bin and say "oh well ya win some ya lose some"...do you have any more suggestions as what to try please? as all of the above still stands except the signature error is "avrdude: Device signature = 0xffffff"

Regards 
Andy

MCUdude, 2019-09-02:
I believe it's caused by your sketch. Somehow the CPU enters power-down mode, and then you're not able to communicate with it until you do a hard reset. However, it doesn't take many milliseconds before the CPU is powered down, and we're back again.

When I _really_ screw up something I usually pull out my old STK500 board and perform some high voltage parallel programming. HVPP always works no matter how bad you messed up.

I don't know what more I can do. I'm pretty confident it's caused by the program that's running. My suggestion would be to be very careful with the power-down mode. If you need more guidance I suggest you try over at the Avrfreaks forum. They are very good at this and probably knows what's really causing the problem and how power.h should be used to prevent situations like these. Again, I have very little experience with power.h, so I can't help you much further I'm afraid.

I'll try to upload the same sketch to a 328P I have. If I'm totally screwed I can always use my STK500 to recover it. I'll keep you updated!

andymouse123, 2019-09-02:
Thank you so much!

MCUdude, 2019-09-07:
I got some spare time this evening, so I loaded your sketch over to an Arduino UNO. I'm always able to re-upload the sketch using the regular USB to serial interface. I'm also able to re-burn the bootloader using a USBasp programmer.

> If it helps any, I just realized that its running on an INTERNAL clock and if I remember correctly when this all started I was messing around with the prescaler don't no if this is relevant or not but 

Internal or external oscillator shouldn't matter here. It's not possible to "accidentally" disable the reset pin without playing around with the fuses. MiniCore don't let you disable the reset pin. I'm not sure what's wrong really. Most likely your hardware, but it's difficult for me to tell.

andymouse123, 2019-09-08:
Hi
Thanks for your and your colleagues help in this I have still learned a lot. I think we can say that this chip 
 is just "faulty" in some way, and this is just an odd failure mode. I think I will use it to try HV fuse setting and stuff like that as there is nothing to lose, thanks once again for your help.

Regards 

Andy

CGarces, 2019-10-19:
I have the same error
`avrdude: Device signature = 0x000000`
Tested with 3 diferent chips.

I'm using the internal 8MHz clock, with arduino nano as ISP.
Im using the simplified schema without esternal clock as described there:

https://www.instructables.com/id/ATmega8-As-Arduino-using-Internal-8Mhz-Crystal/

d35k, 2019-10-26:
Just dont forget to use 10. pin as reset.

MCUdude, 2020-09-10:
I'm closing this issue because there's nothing wrong with MiniCore that causes this issue.

closed 2020-09-10 by MCUdude