MCUdude/MiniCore #108: Wire1 working incorrectly on 328PB

K4ktus123, 2019-09-06:
I have a sketch that reads data from DS3231 RTC and prints it over serial. It works fine on first I2C port, but when I tried to connect it to second I2C it didn't work. Prints some garbage numbers over serial and freezes. Please let me know if you have any questions. Thanks!

MCUdude, 2019-09-06:
Does the i2c_scanner sketch work on the 328PB when using Wire1.h?

K4ktus123, 2019-09-07:
No, it doesn't. Same thing as with my sketch - prints some garbage and freezes.

MCUdude, 2019-09-07:
I suspect there's something wrong with your chip or something. Here's the output of the i2c_scanner when connected my DS3231 board (whick also has an on-board i2c eeprom) to TWI1.

```
Scanning...
I2C device found at address 0x57  !
I2C device found at address 0x68  !
done
```

K4ktus123, 2019-09-07:
This might be the case, although everything else seems to work fine. (SPI1 and UART1 not tested)
I have a spare chip, but I don't have a second adapter board. I will get one on Monday or Thursday. If you come up anything else, please let me know. I'll post here as soon as get the second chip soldered and running. The weird part is that it behaves a little random, and now I got something like this:
[image]

Maybe a wiring problem?

K4ktus123, 2019-09-07:
I quickly wrote this little sketch:
```
#include <Wire1.h>

void setup() {
  Serial.begin(9600);
  Wire1.begin();
}

void loop() {
  Wire1.beginTransmission(0x68);
  Wire1.write(0x01);
  Wire1.endTransmission();
  Wire1.requestFrom(0x68, 1);
  Serial.println(Wire1.read(), BIN);
  delay(1000);
}
```
And the output looks like this:
```
11111111111111111111111111111111
1001001
1001001
1001001
1001001
111111111111111111111111
1\1111111111111111
1001001
1\111111111111111111111111111111
1010000
111\111111111111111111111111111
1010000
1010000
1010000
1010000
1111
```

MCUdude, 2019-09-07:
Mine output

```
110011
110011
110011
110011
110011
110011
110011
110011
110011
110011
110011
110011
110011
110011
110011
```

K4ktus123, 2019-09-07:
Just out of curiosity I tried it with internal oscillator, and surprisingly it works. So it must be some oscillator-related problem. How do you have the crystal connected on your board?

MCUdude, 2019-09-07:
I am using a 16 MHz oscillator, but since the 328PB does not have a full swing oscillator driver, it's very touchy compared to other AVRs. It's very sensitive to capacitive and resistive loads compared to 328P. For harsh environments, I'd rather use a dedicated clock to drive the 328PB just to prevent what you've experiencing.

K4ktus123, 2019-09-07:
Ok, I'll see what I can do with that. Maybe some dedicated driver? Is it possible to just feed the clock signal into one of the pins?

MCUdude, 2019-09-07:
Yes, it is. Just select some external oscillator option and burn bootloader. I've made a crystal driver for the ATtiny13, which does not come with a builtin crystal driver at all.

The schematics for it looks like this, where C1 and C2 = 10pF, R1 = 1M and R2 = 680R
[image]

K4ktus123, 2019-09-07:
This is exactly what I needed! Just one more thing: does it matter which crystal pin I connect the signal to?

MCUdude, 2019-09-07:
> Just one more thing: does it matter which crystal pin I connect the signal to?

Yes, it does. You'll have to connect it to XTAL1

K4ktus123, 2019-09-07:
Alright, thank you so much!

closed 2019-09-07 by K4ktus123