MCUdude/MiniCore #110: tone() function problem when i change it slowly

Master811129, 2019-09-08:
hello there i have a little problem with the tone() function.
i wrote this sketch for a siren:
`void setup()
{
  pinMode(9,OUTPUT);
}

void loop()
{
  for(uint16_t i = 300 ;i<1500;i++)
  {
    tone(9,i);
    delay(2);
  }
  for(uint16_t i = 1500 ;i>300;i--)
  {
    tone(9,i);
    delay(2);
  }
}`

it's OK but i hear some "tic tic" noises randomly.
i also changed the code and disabled millis() function because i thought it's for millis() interrupt....

`#include <util/delay.h>

void setup()
{
  TCCR0=0;
  pinMode(9,OUTPUT);
}

void loop()
{
  for(uint16_t i = 300 ;i<1500;i++)
  {
    tone(9,i);
    _delay_ms(2);
  }
  for(uint16_t i = 1500 ;i>300;i--)
  {
    tone(9,i);
    _delay_ms(2);
  }
}`

but this problem still randomly appears. can you help me please?

Master811129, 2019-09-08:
OK it seems i have this problem only with internal tone() function i download this library:
[https://bitbucket.org/teckel12/arduino-new-tone/downloads/](url)
the problem solved when i used this library. plus the sketch complied with only 928bytes!
in my opinion this library should replace with the built-in library.

MCUdude, 2019-09-08:
Well, the official tone library isn't famous for being the best. But MiniCore is supposed to be compatible with the official code, so I won't replace exiting "official" code, even if the code is better, faster, smaller.

If the new tone library makes it into the official Arduino code, I will, of course, add it to MiniCore as well.

Let's just call this a known flaw and a feature 😉

closed 2019-09-08 by MCUdude