MCUdude/MiniCore #111: how to download HEX with MiniCore bootloader

voodooli, 2019-09-23:
I have tried OpenJumper and Arduloader, but they all support official avr bootloader only.
And I need use hex for remote downloading.
Please help me.

MCUdude, 2019-09-23:
What are you trying to accomplish?

voodooli, 2019-09-23:
I wrote my sketch and want to download it for someone else remotely. But I couldn't give him this source code. So hex will be a good choice.

MCUdude, 2019-09-23:
You can get the compiled hex file under Sketch > Export compiled binary.

voodooli, 2019-09-29:
I rearranged my question:
I've compiled my sketch into hex and found it. Then I tried to upload it to my diy Minicore board by FTDI with Avrdudess, Xloader, Openjumper and other softwares/scripts like them. I could upload it by Arduino IDE.
But all failed and it was sad.
Could you please help me to achieve this goal.
Many thanks :)

MCUdude, 2019-09-29:
The picture you edited in was helpful. You won't find MiniCore as a separate option in Avrdudess, because I'm not involved with the development of Avrdudess.

However, If you're running your ATmega328P at 16 MHz, you can just use the Arduino UNO option (baudrate 115200).

voodooli, 2019-09-29:
Hi, I find this:
` <Preset>
    <programmer>arduino</programmer>
    <mcu>m328p</mcu>
    <port>COM1</port>
    <baud>115200</baud>
    <bitclock />
    <flashFile />
    <flashFormat>a</flashFormat>
    <flashOp>w</flashOp>
    <EEPROMFile />
    <EEPROMFormat>a</EEPROMFormat>
    <EEPROMOp>w</EEPROMOp>
    <force>false</force>
    <disableVerify>false</disableVerify>
    <disableFlashErase>false</disableFlashErase>
    <eraseFlashAndEEPROM>false</eraseFlashAndEEPROM>
    <doNotWrite>false</doNotWrite>
    <lfuse>0xFF</lfuse>
    <hfuse>0xDE</hfuse>
    <efuse>0xFD</efuse>
    <setFuses>false</setFuses>
    <lockBits>0xFF</lockBits>
    <setLock>false</setLock>
    <additional />
    <verbosity>0</verbosity>
    <name>Arduino Uno (ATmega328P)</name>`

This code defind a bootloader, and how can I modify it into MiniCore？

MCUdude, 2019-09-29:
> This code defind a bootloader, and how can I modify it into MiniCore？

I don't understand your question. The code above does not define a bootloader or bootloader path.

If you use an 8 MHz external oscillator the baud rate is 57600. All the fuses can be found in the boards.txt file. If you want to get the fuses directly you can just read them out using a programmer and Avrdudess.

voodooli, 2019-09-29:
What about the Bit clock?
And the fuses 0b1111 confused me, I tought it should be 4 figures.
| |
李柏良
|
|
voodooli@163.com
|
签名由网易邮箱大师定制

MCUdude, 2019-09-29:
The bitclock is the speed the programmer uses when it communicates with the target. It's not used when you program a target using a bootloader.

The fuses in boards.txt are represented in binary to gain more control of each individual bit.

This line sets the efuse. {bootloder.cfd_bit} is a placeholder for, you guessed it, the CFD bit. This is defined another place in the file.
```
328.menu.BOD.2v7.bootloader.extended_fuses=0b1111{bootloader.cfd_bit}101
```

If the cfd_bit is 0, the extended fuse will be 0b11110101. If it is 1, the extended fuse will be 0b11111101.

closed 2019-11-16 by MCUdude