MCUdude/MiniCore #115: Bootloader-upload frequency seems to be wrong

BlueTiger12, 2019-11-11:
For PlatformIO, the board_upload.speed for the 8 MHz external crystal seem to be 38400 instead of the described 57600.
MiniCore Bootloader (v 2.0.3) flashed by Arduino IDE with Clock set as "8 MHz external".
Code upload from PlatformOI in VS Code only works with "board_upload.speed = 38400".

MCUdude, 2019-11-11:
See #114. The baud rate in boards.txt is actually wrong. It is supposed to be 57600 @ 8 MHz external. This will be fixed in the next release.

BlueTiger12, 2019-11-11:
Thank you very much.
I didn't know, i had to look in the pull requests too, to find all issues.
Do you want to keep this issue open until the next release?

MCUdude, 2019-11-11:
> Do you want to keep this issue open until the next release?

Sure!

MCUdude, 2019-11-25:
A new version of the atmelavr PlatfromIO boards package (2.0.0) have just been released. This release brings lots of new MiniCore related features, so I'll recommend you to have a look in the [PlatfomIO readme](https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md) again.

You can now specify the upload baud rate and make PlatformIO load the correct bootloader based on this value. This means this issue could be closed. But I'd like to hear if you were able to get it right first!

BlueTiger12, 2019-11-27:
Thanks for the update.
Going to try it today or tomorrow.

If i understand it correctly, i can now specify the upload frequency independent of the crystal used?
So for example i could use a upload rate of 115200 on a 8 MHz internal clock?

MCUdude, 2019-11-27:
> If i understand it correctly, i can now specify the upload frequency independent of the crystal used? So for example i could use a upload rate of 115200 on a 8 MHz internal clock?

That's right! Or, you'll have to select one of the available baud rates in [this table](https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md#board_uploadspeed). IMO the new fuses/bootloader feature makes AVR development with PlatformIO much better!

BlueTiger12, 2019-11-27:
That sounds really neat.
I just tried, and was able to upload with a baudrate of 38400 to an atmega328PB with external 8MHz crystal.
The bootloader (from MiniCore) was already flashed through the Arduino IDE.

I didn't get the reprogramming of the bootloader with VS Code and Platformio to work though.
I have to admit, i never did it before, but when i tried "platformio run -t bootloader" as written in [(https://docs.platformio.org/en/latest/platforms/atmelavr.html#id13)], the result was:
`*** Do not know how to make File target `bootloader' (C:\Users\bluetiger\Documents\PlatformIO\Projects\LivingRoom_Environment\bootloader).  Stop.`

When i then compared the board .json files from the UNO example:
[https://github.com/platformio/platform-atmelavr/blob/develop/boards/uno.json]
with the one for the MiniCore ATmega328PB
[https://github.com/platformio/platform-atmelavr/blob/develop/boards/ATmega328PB.json]
i saw no bootloader section.
Is it missing, or am i just doing it wrong?

Btw., i saw all your bootloader files for the different speeds and options. You really don't stop part-ways, do you? :D
It's an amazing work you did with these different cores. Thank you very much for them.

MCUdude, 2019-11-27:
> I didn't get the reprogramming of the bootloader with VS Code and Platformio to work though.

You will have to upgrade to the latest Atmel AVR package (2.0.0). You can find it on the VSCode PlatformIO home page. Could you also post the contents of the platformio.ini file? It might need some tweaks.

> When i then compared the board .json files from the UNO example:
https://github.com/platformio/platform-atmelavr/blob/develop/boards/uno.json
with the one for the MiniCore ATmega328PB
https://github.com/platformio/platform-atmelavr/blob/develop/boards/ATmega328PB.json
i saw no bootloader section.
Is it missing, or am i just doing it wrong?

I'm not 100% sure how it's done under the hood, but the latest Atmel AVR package will actually download MiniCore as a dependency when using a "MiniCore target" such as ATmega328PB.

> Btw., i saw all your bootloader files for the different speeds and options. You really don't stop part-ways, do you? :D
It's an amazing work you did with these different cores. Thank you very much for them.

Yeah, I started out with MightyCore, where I needed more bootloader files to provide. After creating a bootloader building script (needed to learn bash and awk) I figured it's fairly easy to build for other targets as well. It also turned out that the Arduino core files didn't need much re-work to support other targets as well. That's how MiniCore was born! As of today, all bootloaders are hosted in a [separate repo](https://github.com/MCUdude/optiboot_flash), and the same goes for the [Arduino core files](https://github.com/MCUdude/MCUdude_corefiles). It's nice to hear you find my work useful! It's all just for fun; I don't make any money out of it 😉

BlueTiger12, 2019-11-28:
> You will have to upgrade to the latest Atmel AVR package (2.0.0). You can find it on the VSCode PlatformIO home page. Could you also post the contents of the platformio.ini file? It might need some [...]
I completely missed this update. VS Code showed no indication, but updated now.

> I'm not 100% sure how it's done under the hood, but the latest Atmel AVR package will actually download MiniCore as a dependency when using a "MiniCore target" such as ATmega328PB.
On the first "platformio run -t bootloader" i saw the download.

I think the remaining problems will be my incomplete .ini file
It is mostly taken from your example, but i am unsure, how to modify it for bootloader upload with and arduino uno as isp
Here it is:
`; ENVIRONMENT SETTINGS
[env:MiniCore]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = ATmega328PB
; Clock frequency in [Hz]
board_build.f_cpu = 8000000L

; HARDWARE SETTINGS
; Oscillator option
board_hardware.oscillator = external
; Hardware UART for serial upload
board_hardware.uart = uart0
; Brown-out detection
board_hardware.bod = disabled
; EEPROM retain
board_hardware.eesave = yes

; UPLOAD SETTINGS
board_upload.speed = 38400
; Upload serial port is automatically detected by default. Override by uncommenting the line below
;upload_port = /dev/cu.usbserial*

; BUILD OPTIONS
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
build_flags = 

; Upload using programmer
;upload_protocol = usbasp
; Aditional upload flags
;upload_flags = -Pusb

; SERIAL MONITOR OPTIONS
; Monitor and upload port is the same by default
;monitor_port = 
; Serial monitor baud rate
monitor_speed = 115200`

The current log is:
`$ platformio run -t bootloader
Processing MiniCore (platform: atmelavr; framework: arduino; board: ATmega328PB)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega328PB.html
PLATFORM: Atmel AVR 2.0.0 > ATmega328PB
HARDWARE: ATMEGA328PB 8MHz, 2KB RAM, 31.50KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduino-avr-minicore 2.0.3
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MySensors> 2.3.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|   |-- <EEPROM> 2.0
Building in release mode
Target configuration:
Target = atmega328pb, Clock speed = 8000000L, Oscillator = external, BOD level = disabled, UART port = uart0, Save EEPROM = yes
Selected fuses: [lfuse = 0xff, hfuse = 0xd6, efuse = 0xf7]
Setting fuses
avrdude: ser_open(): can't open device "unknown": The system cannot find the file specified.

avrdude done.  Thank you.

*** [bootloader] Error 1
====================================================================== [FAILED] Took 1.57 seconds ======================================================================
`

Thank you very much for your help :-)

BlueTiger12, 2019-11-28:
I got it to work with the addition of the following in the platform.ini file:
```
upload_protocol = stk500v1
; each flag in a new line
upload_flags =
    -P$UPLOAD_PORT
    -b$UPLOAD_SPEED

; edit these lines
;upload_port = COM4
;upload_speed = 19200
```

upload_speed 38400 and 57600 did not work. (not in sync)

After burning the bootloader, programming was only possible with an upload_speed of 19200

I think i am still missing something here

MCUdude, 2019-11-28:
> upload_speed 38400 and 57600 did not work. (not in sync)

I'm not really surprised by this. The internal oscillator is often very inaccurate and isn't really a good choice when you need accurate serial communication. How does 250000 baud work? This baud rate has significantly less error on 8 MHz, so you have a better chance of getting it to work. If you want to know more about baud rates, clocks and error I suggest you have a look at some [calculated baud rate tables for AVRs](https://trolsoft.ru/ru/uart-calc). Note that the bootloader using the double speed option, so look at the right part of the table (U2Xn=1) to get the right values.

I once had to scrap a prototype of a commercial product because I used the internal oscillator instead of an external one. The board was communicating at 1200 baud, but even a low baud rate like this caused trouble. I could, of course, have calibrated the internal oscillator, but I figured an external one is so cheap anyway.

BlueTiger12, 2019-11-28:
Good Morning,

i agree with the unstable internal oscillator, on one of my nodes, the supposed 1 hour interval is nearly an 1:30 hour interval on the internal oscillator.

But in this case, i tend to disagree, and think rather, that it is a configuration problem.
The board i used for testing is using an external 8 MHz oscillator and is running on 3.3 Volt.
I have used this already on the 38400 baudrate. The first Generation of my boards uses the internal oscillator and has also worked at 38400 baud.
I previously flashed the bootloader using the arduino ide with you MiniCore.
But atm, with VS Code and Platformio, i can not even flash the Bootloader via ISP (using an arduino Uno as ISP) with this Baudrate. Only with 19200.

MCUdude, 2019-11-28:
Maybe you could post the entire verbose output from PlatformIO? you can do this by running 
```pio run -t -v```

I don't have much experience with Arduino as ISP. I absolutely prefer to use a dedicated programmer such as the USBtinyISP or USBasp.

BlueTiger12, 2019-11-28:
Platformio.ini:
```
[env:MiniCore]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
board = ATmega328PB
board_build.f_cpu = 8000000L

; HARDWARE SETTINGS
board_hardware.oscillator = external
board_hardware.uart = uart0
board_hardware.bod = disabled
board_hardware.eesave = yes

; UPLOAD SETTINGS
;board_upload.speed = 19200
board_upload.speed = 38400
build_unflags = -flto

build_flags = 
monitor_speed = 115200

upload_protocol = stk500v1
upload_flags =
    -P$UPLOAD_PORT
    -b$UPLOAD_SPEED

upload_port = COM4
upload_speed = 19200
;upload_speed = 38400
```

Log:
```
$ platformio run -v -t bootloader
Processing MiniCore (platform: atmelavr; framework: arduino; board: ATmega328PB; board_build.f_cpu: 8000000L; board_hardware.oscillator: external; board_hardware.uart: uart0; board_hardware.bod: disabled; board_hardware.eesave: yes; board_upload.speed: 38400; build_unflags: -flto; build_flags: ; monitor_speed: 115200; upload_protocol: stk500v1; upload_flags: -P$UPLOAD_PORT, -b$UPLOAD_SPEED; upload_port: COM4; upload_speed: 19200)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega328PB.html
PLATFORM: Atmel AVR 2.0.0 > ATmega328PB
HARDWARE: ATMEGA328PB 8MHz, 2KB RAM, 31.50KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduino-avr-minicore 2.0.3
avr-g++ -o "C:\Users\bluetiger\Documents\PlatformIO\Projects\MoistureTest_Outside\src\Test_Moisture.ino.cpp" -x c++ -fpreprocessed -dD -E "C:\Users\BLUETI~1\AppData\Local\Temp\tmpwout4ca3"
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MySensors> 2.3.1 (C:\Users\bluetiger\.platformio\lib\MySensors_ID548)
|   |-- <Wire> 1.0 (C:\Users\bluetiger\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire)
|   |-- <SPI> 1.0 (C:\Users\bluetiger\.platformio\packages\framework-arduino-avr-minicore\libraries\SPI)
|   |-- <EEPROM> 2.0 (C:\Users\bluetiger\.platformio\packages\framework-arduino-avr-minicore\libraries\EEPROM)
Building in release mode
Target configuration:
Target = atmega328pb, Clock speed = 8000000L, Oscillator = external, BOD level = disabled, UART port = uart0, Save EEPROM = yes
Selected fuses: [lfuse = 0xff, hfuse = 0xd6, efuse = 0xf7]
avrdude -p atmega328pb -C C:\Users\bluetiger\.platformio\packages\tool-avrdude\avrdude.conf -c stk500v1 -PCOM4 -b19200 -e -Ulock:w:0x0f:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xff:m -Uefuse:w:0xf7:m

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

[... 125 lines cut ...]
|   |-- <Wire> 1.0 (C:\Users\bluetiger\.platformio\packages\framework-arduino-avr-minicore\libraries\Wire)
|   |-- <SPI> 1.0 (C:\Users\bluetiger\.platformio\packages\framework-arduino-avr-minicore\libraries\SPI)
|   |-- <EEPROM> 2.0 (C:\Users\bluetiger\.platformio\packages\framework-arduino-avr-minicore\libraries\EEPROM)
Building in release mode
Target configuration:
Target = atmega328pb, Clock speed = 8000000L, Oscillator = external, BOD level = disabled, UART port = uart0, Save EEPROM = yes
Selected fuses: [lfuse = 0xff, hfuse = 0xd6, efuse = 0xf7]
avrdude -p atmega328pb -C C:\Users\bluetiger\.platformio\packages\tool-avrdude\avrdude.conf -c stk500v1 -PCOM4 -b38400 -e -Ulock:w:0x0f:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xff:m -Uefuse:w:0xf7:m
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x66
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x66
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x66
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x66
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x66

avrdude done.  Thank you.

*** [bootloader] Error 1
====================================================================== [FAILED] Took 13.05 seconds ======================================================================
```

MCUdude, 2019-11-28:
Ah, I understand. The Arduino as ISP programmer is a bit confusing.

The Arduino as ISP only supports 19200 baud for burning the bootloader. After the bootloader is burned, you may use a "regular" USB to serial adapter for uploading. That's what the bootloader is for.

In theory, PlatformIO should use `upload_speed` when setting the programmer (Arduino as ISP) baud rate, while `board_upload.speed` is used to set the upload board rate when using the bootloader. Somehow PlatformIO is interpreting these two parameters as the same thing. I will report this issue to the pio devs. This messes up the bootloader selection and causes the same 19200 baud bootloader to be selected ever time.

BlueTiger12, 2019-11-28:
This makes sense.
Sorry, i didn't think of mentioning, that the Arduino as ISP only burns the bootloader and the remaining uploading is done via serial communication.

I can also confirm, that i was able to burn the Bootloader with 38400 baudrate by specifying it manually. Uploading via an USB to serial converter worked too then.
`board_bootloader.file = C:\Users\USERNAME\.platformio\packages\framework-arduino-avr-minicore\bootloaders\optiboot_flash\bootloaders\atmega328pb\8000000L\optiboot_flash_atmega328pb_UART0_38400_8000000L.hex`

Shall i leave this issue open again?
The title is still correct, just the problem is different :D

>I don't have much experience with Arduino as ISP. I absolutely prefer to use a dedicated programmer such as the USBtinyISP or USBasp.
I tried it, because i had no oder compatible programmer at hand, and then stayed with it, mostly because of lazyness, but it also works pretty well, i have to admit.

curie31, 2020-05-25:
Hi all.

I am getting problem burning the bootloader on an ATmega328PB chip using Minicore. This issue seemed relatable, so thought of writing in here.
I am using a 16 MHz crystal as my clock source and following the steps as mentioned on, https://github.com/MCUdude/MiniCore
I am using 'Arduino as ISP' as my programmer. When hitting the 'Burn bootloader' option, i am getting the following error,
`Arduino: 1.8.12 (Windows 10), Board: "ATmega328, Yes (UART0), 328PB, BOD 1.8V, LTO disabled, External 16 MHz"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\hp\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf -v -patmega328pb -cstk500v1 -PCOM8 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0b11110110:m -Uhfuse:w:0xd6:m -Ulfuse:w:0b11111111:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\hp\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
[... 114 lines cut ...]
avrdude: reading input file "C:\Users\hp\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex"
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% -0.00s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against C:\Users\hp\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex:
avrdude: load data flash data from input file C:\Users\hp\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex:
avrdude: input file C:\Users\hp\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex contains 32768 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x7e00
         0x00 != 0x01
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.`

I have tried things like, changing the baud rate in the boards.txt file, using the internal oscillator ( 8MHz as well as 1 MHz), reopening the IDE, etc. but getting the same error.
Also, was thinking to burn the fuse bits as mentioned in the boards.txt file for the 16MHz crystal, is that a safe option?
Any help would be appreciated. Thank you in advance!

MCUdude, 2020-06-21:
This issue was fixed a while ago in https://github.com/platformio/platform-atmelavr/issues/178
Use `board_bootloader.speed = 38400` to set the bootloader baud rate.
Use `board_upload.speed = 19200` to set the upload speed your programmer requires. In this cause, the Arduino as ISP requires 19200 baud to function. After the bootloader is loaded, you'll have to change this field to 38400 to match the bootloader speed.

closed 2020-06-21 by MCUdude