MCUdude/MiniCore #116: MacOS Catalina issues recognizing miniCore

DDeFilippis, 2019-11-11:
I am getting this error with the miniCore setup ATmega328, 16MHz External Clock, BOD 1.8v, LTO Disabled, 328P/328PA, Bootloader (YES UART0)

fork/exec .../Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-g++: bad CPU type in executable
Error compiling for board ATmega328.

I'm not sure if this is related to the issues surrounding Apple dropping 32bit support but this seemed to come with the OS upgrade.

MCUdude, 2019-11-11:
> I'm not sure if this is related to the issues surrounding Apple dropping 32bit support but this seemed to come with the OS upgrade.

I believe it is. IIRC avr-gcc 7.3.0 works with Catalina. To get this compiler, try to Install the Arduino boards package that includes support for the Arduino Wifi Rev2 and the Arduino Nano Every.

DDeFilippis, 2019-11-11:
Shoot I was on 1.6.26 and updating to 1.8.5 did the trick. Thank you for the incredibly fast reply.

The board package is titled "Arduino megaAVR Boards"

closed 2019-11-11 by DDeFilippis