MCUdude/MiniCore #117: [168PB/328PB] Looks like its missign some power AVR functions.

hitech95, 2019-11-18:
So far I have found that thoose functions are missing:
```
  power_spi_disable();
  power_twi_disable();
```

Is there an alternative? Is the Arduino AVR GCC outdated?

Unfortunatly I need `Start of frame IRQ` and only the new PB series have it!

MCUdude, 2019-11-18:
You would probably have to include power.h

``` c
#include <avr/power.h>
```

What IDE version are you using? If the power.h file is somewhat outdated, there's not much I can do about that really. power.h is a part of avr-libc, which I don't have any control over.

hitech95, 2019-11-18:
> 
> 
> You would probably have to include power.h
> 
> ```c
> #include <avr/power.h>
> ```
> 
> What IDE version are you using? If the power.h file is somewhat outdated, there's not much I can do about that really. power.h is a part of avr-libc, which I don't have any control over.

Yea I have imported that, if i switch to a 328p instead of a 328PB it compiles without any trouble,
I was also supecting of an outdated avr-libc. Any idea on how to solve that?

I'm using the Arduino 1.8.10.

After fiew more tests the 168PB don't have that isse but the 328PB yes.
I actually don't know how to troubleshoot this (I'm not an AVR expert)

MCUdude, 2019-11-18:
If you look in the Arduino source files, you will find the power.h file. 
The path on my Mac is
`/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/power.h 
`.

If you look at line 1362 you can clearly see that ATmega328PB is missing (ATmega324PB also).
If you want this to be fixed I suggest opening a new issue over at the [Arduino AVR toolchain repo](/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/power.h 
).

hitech95, 2019-11-18:
> If you want this to be fixed I suggest opening a new issue over at the Arduino AVR toolchain repo.
Link looks broken. Is it this: [toolchain-avr](https://github.com/arduino/toolchain-avr)

closed 2019-11-18 by hitech95

MCUdude, 2019-11-18:
As a temporary fix you can just add ATmega328PB to power.h. However, this needs to be fixed in the toolchain, so that you don't have to manually do this every time you update your IDE.

hitech95, 2019-11-18:
> 
> 
> As a temporary fix you can just add ATmega328PB to power.h. However, this needs to be fixed in the toolchain, so that you don't have to manually do this every time you update your IDE.

I have tried but I still get the error, I have added annoder case after the 328p, but I still get the error.
Maybe there is a cache or someting somewhere.