MCUdude/MiniCore #118: Shematics

EliasA97, 2019-11-19:
Hi, sorry for my off-topic question but I didn't find anywhere else to ask. Is it possible to make a basic schematic for ATSAM3X8E like the ones you've already made?

MCUdude, 2019-11-19:
I guess you're referring to the minimal setup schematics? I have never used an ATSAM3X8E in a project before, and have no idea what a minimal setup for this would look like. The only board I know of that uses the 3X8E is the Arduino Due. You can always have a look in its [schematic](https://content.arduino.cc/assets/DUE_V02g_sch.pdf) if that helps.

closed 2019-11-19 by MCUdude

EliasA97, 2019-11-19:
Thank's for the reply. I recently made a working atmega2560 board with your schematic. The thing is that your schematics are simple enough and that's why they're helpful. On the other hand the official arduino schematics are too complicated for me at the moment, but unfortunatelly I'll have to stick with them. 
Thank's again!!

MCUdude, 2019-11-19:
I do agree that many (if not all!) of the official Arduino schematic is just messy and complicated for what it is. That's why I provided my own schematics in the first place.

While searching for a better Arduino DUE schematic, I found this; a DUE compatible board with a bare minimum of components. The schematic is definitely much easier to read.
https://forum.arduino.cc/index.php?topic=275705.0

EliasA97, 2019-11-19:
Well, it's weird how I missed it but it probably is what I want. Thank's for your effort!