MCUdude/MiniCore #121: Can't compile "return;" statement.

makstrix, 2019-12-15:
On MiniCore I can't compile program which contains "return;"
It gives an error:
`error: return-statement with no value, in function returning 'uint8_t {aka unsigned char}' [-fpermissive]`
on Optiboot and Arduino bootloaders chosen it is only a warning and compiles with no error:
`warning: return-statement with no value, in function returning 'uint8_t {aka unsigned char}' [-fpermissive]`
Please help me, I'm a noob :)

MCUdude, 2019-12-15:
How about adding a value to that return, such as `return 0;`?

makstrix, 2019-12-15:
Wow @MCUdude you are so fast!
It compiles without any error, but the program is bigger and I would like it to stay with "return;"
Is it possible?

MCUdude, 2019-12-15:
> Wow @MCUdude you are so fast!

I was just writing on another issue when this popped up.

> It compiles without any error, but the program is bigger and I would like it to stay with "return;"
Is it possible?

Doing a return without a return value is bad practice. MiniCore is a bit more picky on bad code practices than the official Arduino cores. To fix this you can add the -fpermissive flag to platform.txt. You can read more about what it is and what is does [here](https://stackoverflow.com/questions/8843818/what-does-the-fpermissive-flag-do).
Bottom line: You do have an error in your code. Adding -fpermissive just turns this error into a warning instead. You still have an error there, the only difference is that the compiler doesn't complain. If you need to make the program smaller I recommend you to optimize in other ways and/or running on LTO from the Tools menu.

makstrix, 2019-12-15:
Thank you! That worked.

closed 2019-12-15 by MCUdude