MCUdude/MiniCore #122: I2C slave does not work above MiniCore v1.0.4

evansgl, 2019-12-19:
Seems that all versions of MiniCore v2.x.x are affected by the same issue.
I have an atmega328 as an I2C slave and it does not work properly.
It does not report anything in the I2C channel.
If I use the stock and latest arduino hardware pack the I2C slave works. It also works fine with any 1.x.x version of Minicore.

MCUdude, 2020-01-03:
What IDE version are you using? From v2.0.0 and later MiniCore hosts its own version of Wire, while v1.0.4 uses the version that ships with the IDE.

Does it work in your current IDE when using the Arduino UNO as target?

evansgl, 2020-01-05:
Arduino IDE is 1.8.10
I2C Slave works perfectly fine when I use UNO as target.
Regards,
Evans

MCUdude, 2020-01-05:
Thanks! I compared the two versions, and there were done some changes to Wire.cpp in 2017 (to fix slave functionality) that I didn't catch. I'll fix it soon! Would you test it for me after I've pushed a fix? 🙂

MCUdude, 2020-01-05:
@evansgl I've now pushed a fix. Could you test it with your hardware? You'll have to do a manual installation of MiniCore to get the latest fix.

evansgl, 2020-01-05:
Thanks Hans,
Let me try it and I'll let you know tomorrow.
Regards,
Evans

MCUdude, 2020-01-10:
Did you get the chance to test this? 🙂

evansgl, 2020-01-11:
Apologies for the delay.
Unfortunately I tried yesterday but it didn't work.
Copied the latest submitted Wire libraries from git and cleaned all my
built files.
Recompiled it and the atmega is still not working as a i2c slave.
To further test, I copied the original Wire library in the Minicore folder
and this worked.
To be honest I didn't check the code you submitted yet. I just moved the
two Wire* libraries into my workspace
Regards,
Evans

MCUdude, 2020-01-11:
That's strange. Did you only copy the Wire.h and Wire.cpp? You also need the two files (twi.h and twi.c) located in the utility folder.

evansgl, 2020-01-11:
I have just cloned the git MiniCore repo and fully replaced "2.0.3" library
directory. I also verified that the compiler was using the correct path.
Unfortunately it does not work.

-I/Users/evansg/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.3/libraries/Wire/src
/Users/evansg/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.3/libraries/Wire/src/Wire.cpp
-o
/var/folders/56/h22y8dw13mz50028k3z4381c0000gn/T/arduino_build_890922/libraries/Wire/Wire.cpp.o
/Users/evansg/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc
-c -g -Os -std=gnu11 -ffunction-sections -fdata-section

MCUdude, 2020-01-11:
OK. I will try it myself on my own hardware to see if I can figure it out

MCUdude, 2020-01-15:
I did test the master/slave functionality using this code:

https://www.arduino.cc/en/Tutorial/MasterWriter

I used a 328P as a slave. It works just fine with the fix I applied.

mhmayyan, 2020-02-01:
I am using Atmega328PB, MiniCore 2.0.3, Arduino IED 1.8.10.
A slave sender detects a request from a master reader but does not respond. It does not send the bytes (Wire.write()). I am testing the examples "master_reader" and "slave_sender". I used a digital analyzer to see the SDA, SCL lines. I can only see the request from the master but there is no response from the slave.

The two ends are Atmega328PB.

It worked when I replaced the Wire directory from libraries in MiniCore by the Wire folder from standard Arduino AVR core 1.8.2 libraries.

I appreciate your huge efforts. Thank you very much.

MCUdude, 2020-02-01:
@mhmayyan can you retry your test but with a manual install of MiniCore instead? I did push a fix for this that solved this issue for me at least.

mhmayyan, 2020-02-01:
I deleted all the contents of the folder "..\MiniCore\hardware\avr\2.0.3". Then I downloaded the zip file of the repository and extracted the contents of "MiniCore-master.zip\MiniCore-master\avr" into "..\MiniCore\hardware\avr\2.0.3".

The problem is now back. The slave sender does not respond to the master reader but it like before still detects the request from the master reader.

Again, when I replaced the Wire folder in libraries by the one from Arduino AVR it worked.

MCUdude, 2020-02-01:
That's really strange. It works for me!

Can you turn on the compiler output, and have a look at what SPI.cpp file is being compiled, open this and check if [this fix](https://github.com/MCUdude/MiniCore/commit/3be80b64d63618413f00b6e40873607ee2b99602) are applied to the file?

MCUdude, 2020-02-01:
I compared the source files from IDE 1.8.10 with the files from MiniCore, and there was some code lines in the MiniCore version that shouldn't have been there. These have been removed now.

mhmayyan, 2020-02-02:
Now it works.

Thank you very much.

MCUdude, 2020-02-02:
Thanks for testing! Now the library _is_ identical to the official one too. I wil apply this fix to my other Arduino cores as well.

closed 2020-02-02 by MCUdude