MCUdude/MiniCore #123: LGT8F328P support?

esp13, 2019-12-25:
Hi,

I wonder to know if the atmega 328P clones LGT8F328P are already / will be supported by minicore?
If yes, is it / will it possible to do same low power consumption configuration? (lower frequency, disable/lower brown out detection, disable ADC...)

MCUdude, 2019-12-25:
I do have an LGT8F328P board, and I did infact spend some time with it to figure out if I should add it to Minicore or not.

I decided that it wasn't worth it. It seems like it has a custom programming interface that requires a custom version of the Arduino as ISP sketch. That means you can't use a standard programmer. The documentation for the chip isn't very good. Or, I haven't been able to find proper documentation for it.

closed 2019-12-25 by MCUdude

esp13, 2019-12-26:
Damn! The chip seems to have some more advantages against atmel328p and is cheaper.
Anyway thanks for the answer :)

If I find a good/full documentation I will came back with the link, just in case of.

PS: sorry for my poor english level