MCUdude/MiniCore #124: ATMega8 at 9600 with internal oscillator not uploading

sjlongland, 2019-12-30:
Hi,

I'm trying to get a boot-loader going on a little ATMega8L I had laying around.  (The ATMega8L is basically the same as the ATMega8, but limited to 8MHz and an operating voltage range of 2.7-5.5V.  (The original project I bought these for called for 3.3V.)

I'm doing this outside the Arduino/PlatformIO environment under Linux.  Ultimately this chip will be talking to a Raspberry Pi over a UART link (possibly RS-232 using MAX232 at the ATMega8L end, they'll be ~1.5m apart, so too far for SPI) and I want to allow field re-programming of the device without needing ICSP.

I managed to get the boot-loader flashed, having tried both building it myself and using the pre-built images, can't remember which one worked in the end as I was literally trying every trick in the book I could think of in the absence of documentation.  The fuse settings was the most difficult bit to deduce, and I think I found high=`0xe1` low=`0xd4` worked after much reading of the datasheet and reverse-engineering `boards.txt`.

Eventually, I got the LED on `PB5` blinking, so the boot-loader is active, and `avrdude` is able to read the chip signature, so *something* is alive.  The chip is sitting on a breadboard with supply via a LM7805 connected to 7.5V battery supply, and is putting out just over 5V.  I have 100µF capacitance across the VCC/GND pins.  Both GND pins connect to 0V.

Alas, no go trying to actually flash the MCU:

```
RC=0 stuartl@rikishi /tmp $ avrdude -p atmega8 -b 9600 -c arduino -P /dev/ttyUSB0 -U flash:w:uart.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9307 (probably m8)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "uart.hex"
avrdude: writing flash (192 bytes):

Writing | ################################################## | 100% 0.29s

avrdude: 192 bytes of flash written
avrdude: verifying flash memory against uart.hex:
avrdude: load data flash data from input file uart.hex:
avrdude: input file uart.hex contains 192 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.29s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x12
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:00, L:00)

avrdude done.  Thank you.
```

The fact that it reads back as all ones tells me maybe I forgot a setting in the fuses.

Is there some trick to flashing this directly with `avrdude`?

MCUdude, 2020-01-03:
Your high and low fuses look correct. However, it may be your lock bits that prevent the chip from being written to. Did you burn the bootloader using MiniCore and Arduino IDE? It may also be the internal oscillator that isn't accurate enough and needs a tune. Search google for AVR OSCCAL for more info.

You can flash the chip directly using the Avrdude command you've posted:
`$ avrdude -p atmega8 -b 9600 -c arduino -P /dev/ttyUSB0 -U flash:w:uart.hex:i`

sjlongland, 2020-01-04:
I don't have the Arduino IDE installed… it was programmed directly using `avrdude` and a STK500v2 clone programming dongle.  I'll check the lock bits, but those should be at factory defaults.

MCUdude, 2020-01-04:
I'd check the lock bits. Seems like the bootloader doesn't have access to the flash memory.

MCUdude, 2020-01-25:
Did you figure it out?

sjlongland, 2020-01-30:
Hi, haven't had a chance at this stage… currently I'm in hospital and I think the nurses would freak if they saw a bare ATMega8L in the wild (even though there's probably quite a few, and other MCUs, hiding in things they use every day).

closed 2021-06-06 by MCUdude