MCUdude/MiniCore #125: OSCCAL sketch

MCUdude, 2020-01-03:
It would be nice to have an OSCCAL sketch similar/identical to the one MicroCore has. @nerdralph would it be possible to make one that uses the default serial pins, so that the sketch can run on other platforms (MightyCore, MegaCore etc.) as well without modifications?

nerdralph, 2020-01-04:
That shouldn't be too hard, at least for the megaxx8 series.  I'm not aware of any avr-libc define for the RXD/TXD pins, but I think it's always PD0/PD1.

For reliable serial speed on the megaxx8 devices, I've been mulling over an idea to use picobootArduino's autobaud feature.  I could make a new version which saves the bootloader UBRR value in the bootloader flash space, and loads it into UBRR after every reset before starting the application code.
You would then modify Serial.begin(void) to only turn on the USART, leaving UBRR and U2X at their default values set by the bootloader.
If you haven't looked at picobootArduino before, here's my blog post about it.
http://nerdralph.blogspot.com/2018/05/picoboot-adruino-with-autobaud.html

MCUdude, 2020-01-04:
> That shouldn't be too hard, at least for the megaxx8 series. I'm not aware of any avr-libc define for the RXD/TXD pins, but I think it's always PD0/PD1.

One solution would be to provide a separate file, osccal.h where everything hardware specific is defined. This would make it much more elegant if we're trying to get it to work on other AVR families as well.

I'm simply not competent enough to build something like this myself though. At least not the core functionality. However, I'm absolutely capable of making a nice sketch around a proof-of-concept 🙂 

> For reliable serial speed on the megaxx8 devices, I've been mulling over an idea to use picobootArduino's autobaud feature. I could make a new version which saves the bootloader UBRR value in the bootloader flash space and loads it into UBRR after every reset before starting the application code.
You would then modify Serial.begin(void) to only turn on the USART, leaving UBRR and U2X at their default values set by the bootloader.

Does this mean it's possible to add autobaud functionality to Optiboot? If so, that's an interesting thought!

nerdralph, 2020-01-04:
> Does this mean it's possible to add autobaud functionality to Optiboot? If so, that's an interesting thought!

Possible, but but I don't see any point in doing that when I've already implemented it in picobootArduino.  It's half the size of Optiboot, compatible with Arduino just like Optiboot, and doesn't use WDTR (so MCUSR isn't messed up).  Why would you still want to use Optiboot?

MCUdude, 2020-01-04:
I'm using Optiboot because it's easily maintainable because it's written in C. It's also been proved to be extremely reliable, and it's a source code I know. Only devices with 16 kB and less can have a boot section size of 256 bytes.

But, I will download your source code and play around with it 🙂 

 But an OSCCAL sketch would still be great.

MCUdude, 2020-11-20:
Not going to bother with this. If an OSCCAL calibrator was created, we would also have to modify the bootloader to check if an OSCCAL value was stored in EEPROM.

closed 2020-11-21 by MCUdude