MCUdude/MiniCore #129: 4MHz possible?

smartroad, 2020-03-15:
Hi! I was wondering if there is anyway to set the compiler to do 4MHz oscillator? I ask as I am making a IR remote control and wanted it to have it able to run on 2 AAA rechargeable batteries.

MCUdude, 2020-03-15:
That is certainly possible, and there are available bootloaders for 4 MHz. I assume you will be using an external crystal?

You will have to modify boards.txt though. Add a new section based on this one:
https://github.com/MCUdude/MiniCore/blob/fa9e70f1979ae1b68ca772eb15fce2d4943df746/avr/boards.txt#L155-L158

All you need to do is to change f_cpu and baud rate. You can find all available baud rates here:
https://github.com/MCUdude/MiniCore/tree/master/avr/bootloaders/optiboot_flash#precompiled-binaries

For 4 MHz I can recommend 250000 baud.

MCUdude, 2020-03-15:
You can use this. Just replace the prefix on every line (328) with the chip number you're using.

```
 328.menu.clock.4MHz_external=External 4 MHz 
 328.menu.clock.4MHz_external.upload.speed=250000 
 328.menu.clock.4MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits} 
 328.menu.clock.4MHz_external.build.f_cpu=4000000L 
```

Danixu, 2020-04-10:
Hello,

Is possible to add 4 Mhz also using internal clock?. 8Mhz consumption is good, and 1Mhz is incredible, but I'm having problems with 1Mhz clock speed, so surely will work better with 4Mhz.

Thanks!!

MCUdude, 2020-04-10:
> Is it possible to add 4 Mhz also using internal clock?. 8Mhz consumption is good, and 1Mhz is incredible, but I'm having problems with 1Mhz clock speed, so surely will work better with 4Mhz.

That is certainly possible, but it requires a bit of work. Maybe I'll have it ready in the next release?

Danixu, 2020-04-10:
Hello,

Thank you very much!!. I'll wait until then.

Best regards.

MCUdude, 2020-04-11:
I've just pushed some code that lets you use the internal oscillator at 2 or 4 MHz. It's not available in the boards manager release yet, so you will have to do a manual install.

MCUdude, 2020-04-11:
Oh, I also added support for 4, 2 and 1 MHz external oscillators. Let me know if you experience any issues!

closed 2020-04-11 by MCUdude

Danixu, 2020-04-11:
How fast!!, thanks you very much. I'll take a look ;)

commit ba182077 (2020-04-11): Add support for 4, 2 and 1 MHz external clock
Closes #129