MCUdude/MiniCore #130: Minor miscalculation in delayMicroseconds()

LaZsolt, 2020-03-15:
Hi!

I found a minor miscalculation in wiring.c and delayMicroseconds()

In case of 32 MHz clock speed, at line 201 microseconds multiplied by 8 takes 15 clock cycles. 
`   us <<= 3; // x8 us, = 6 cycles not true`

The assembly code:
`        63 e0      ldi     r22, 0x03 ; 1 clock cycle`
`        88 0f       add  r24, r24    ; 3 x 1 clock cycles`
`        99 1f       adc   r25, r25   ; 3 x 1`
`        6a 95      dec   r22		; 3 x 1`
`	e1 f7       	brne	.-8      	; 2 x 2 + 1`
; Sum 15 Cycles

MCUdude, 2020-03-15:
Thanks for reporting. I don't have time to look into this right now, but do you want to propose a fix, so that timing is accurate (and perhaps test it on actual hardware)?

LaZsolt, 2020-03-16:
Need some tests compilation and disassembling to view the effect of volatile. I think, the solution is:
`void delayMicroseconds( volatile unsigned int us)
{
#if F_CPU >= 32000000L
if (!us) return;

us <<= 1;
nop();    // This NOP will break optimization
us <<= 2;

us -= ?;
...`

LaZsolt, 2020-03-22:
Voliatile variable was not a good idea. Compiler generate too complex and big code.

But there is another problem even not using voliatile variable. If delayMicroseconds() called with constants only compiler generate function call free code. This example is from a complex code using timing in Onewire library.

Need more tricks. Thinking...

`void delayMicroseconds( unsigned int us)`
`{`
`#if F_CPU >= 32000000L`
`  if (!us) return;`
`  us <<= 3;`
`  us -= 3;`
`#endif`
`__asm__ __volatile__ (`
`  "1: sbiw %0,1" "\n\t"`
`  "brne 1b" : "=w" (us) : "0" (us)`
`);`
`}`
`delayMicroseconds(53);`

compiled to:

`d78:	e3 ea       	ldi	r30, 0xA3	; 165`
`d7a:	f1 e0       	ldi	r31, 0x01	; 1`
`d7c:	31 97       	sbiw	r30, 0x01	; 1`
`d7e:	f1 f7       	brne	.-4      	; 0xd7c`

LaZsolt, 2020-05-04:
Here are some compiled and disassembled codes. Link time optimization (LTO)  must be disabled.
```
// 32 MHz
delayMicroseconds(3);       // = 6 cycles
    3302:   83 e0           ldi     r24, 0x03
    3304:   90 e0           ldi     r25, 0x00
    3306:   0e 94 a6 1b     call    0x374c      ; 0x374c <delayMicroseconds>
.
.

void delayMicroseconds(unsigned int us)
{
  if (!us) return;          // = 3 cycles, (4 when true)
    374c:   00 97           sbiw    r24, 0x00
    374e:   41 f0           breq    .+16        ; 0x3762 <delayMicroseconds+0x16>

  us <<= 3;          // x8 us, =15 cycles (instead of 6 cycles)
    3750:   23 e0           ldi     r18, 0x03   ; 1 c
    3752:   88 0f           add     r24, r24    ; 1 c x3
    3754:   99 1f           adc     r25, r25    ; 1 c x3
    3756:   2a 95           dec     r18         ; 1 c x3
    3758:   e1 f7           brne    .-8      	; 2+2+1 cycles
	                                            ; 0x3752 <delayMicroseconds+0x6>

  if (!(us -= 8)) return;   // = 3 cycles, (4 when true)
    375a:   08 97           sbiw    r24, 0x08
    375c:   11 f0           breq    .+4         ; 0x3762 <delayMicroseconds+0x16>

  // busy wait
  __asm__ __volatile__ (
    375e:   01 97           sbiw    r24, 0x01
    3760:   f1 f7           brne    .-4         ; 0x375e <delayMicroseconds+0x12>
  );
 
}                    // return = 4 cycles
    3762:   08 95           ret
```

```
// 24 MHz
delayMicroseconds(3);       // = 6 cycles
    3302:   83 e0           ldi     r24, 0x03
    3304:   90 e0           ldi     r25, 0x00
    3306:   0e 94 a6 1b     call    0x3758      ; 0x3758 <delayMicroseconds>
.
.

void delayMicroseconds(unsigned int us)
{
    3758:   9c 01           movw    r18, r24
  if (!us) return;         // = 1+2 cycles, (4 when true)
    375a:   89 2b           or      r24, r25
    375c:   49 f0           breq    .+18         ; 0x3770 <delayMicroseconds+0x18>

  us *= 6;          // x6 us, = 7+1 cycles
    375e:   46 e0           ldi     r20, 0x06
    3760:   42 9f           mul     r20, r18
    3762:   c0 01           movw    r24, r0
    3764:   43 9f           mul     r20, r19
    3766:   90 0d           add     r25, r0
    3768:   11 24           eor     r1, r1

  us -= 5;                  // = 2 cycles
    376a:   05 97           sbiw    r24, 0x05

  // busy wait
  __asm__ __volatile__ (
    376c:   01 97           sbiw    r24, 0x01
    376e:   f1 f7           brne    .-4          ; 0x376c <delayMicroseconds+0x14>
  );
[... 63 lines cut ...]
    3782:   87 95           ror     r24         ; 1 c x3
    3784:   fa 95           dec     r31         ; 1 c x3
    3786:   e1 f7           brne    .-8         ; 2+2+1 0x3780 <delayMicroseconds+0x48>
    3788:   82 0f           add     r24, r18    ; 1 c
    378a:   93 1f           adc     r25, r19    ; 1 c
    378c:   84 0f           add     r24, r20    ; 1 c
    378e:   95 1f           adc     r25, r21    ; 1 c
  }

  // busy wait
  __asm__ __volatile__ (
    3790:   01 97           sbiw    r24, 0x01
    3792:   f1 f7           brne    .-4      	; 0x3790 <delayMicroseconds+0x58>
  );

}                    // return = 4 cycles 
    3794:   08 95           ret

  else 
  { 
    us -= 8;                // = 4 cycles
    3796:       08 97       sbiw    r24, 0x08
    3798:       fb cf       rjmp    .-10     	; 0x3790 <delayMicroseconds+0x58>
  }
```

LaZsolt, 2020-05-04:
If you don't want to delay zero time and every timing are constans (not variable), this most compact and precise code is for you. (Must be in Arduino.h) LTO does not matter any more,
```
#define delayMicroseconds(us)     \
(__extension__({                  \
  __asm__ __volatile__ (          \
    "usL_%=:" "sbiw %0,1" "\n\t"  \
              "brne usL_%="       \
	          :  /* no outputs */ \
              :"w" ((uint16_t) ((F_CPU/4000000L) * (uint16_t)us))  \
  );                              \
}))
```

LaZsolt, 2020-08-28:
The code above has an error. Higher values of requested microseconds cause compiler 32 bit overflow.
The newer, better version:
````
#define delayMicroseconds(us)     \
(__extension__({                  \
  __asm__ __volatile__ (          \
    "usL_%=:" "sbiw %0,1" "\n\t"  \
              "brne usL_%="       \
              :  /* no outputs */ \
              :"w" ( (uint16_t) (( (F_CPU/1000) * (uint32_t) us ) / 3000L) )  \
  );                              \
}))
````

LaZsolt, 2020-09-09:
I created a new version of delayMicroseconds() function. I would be happy if someone check it: https://github.com/LaZsolt/delayMicroseconds/tree/master/for_MCUdude/function
And I calculated cycle count for every freq: https://github.com/LaZsolt/delayMicroseconds/tree/master/Analysis

MCUdude, 2020-09-09:
Sorry for staying quiet for so long. I'll see if I can look into it tonight.

But thanks for your work! Improvements are always welcome 🙂

MCUdude, 2020-09-10:
so, what would you recommend using?

```c
#define delayMicroseconds(us)     \
(__extension__({                  \
  __asm__ __volatile__ (          \
    "usL_%=:" "sbiw %0,1" "\n\t"  \
              "brne usL_%="       \
              :  /* no outputs */ \
              :"w" ( (uint16_t) (( (F_CPU/1000) * (uint32_t) us ) / 4000L) )  \
  );                              \
}))
```

is of course way shorter, but I'm so inline assembly guy, so I can't really tell what's going on. Will the function above work for all F_CPU values? How about the code provided at https://github.com/LaZsolt/delayMicroseconds/tree/master/for_MCUdude/function?

Speaking of the code here: https://github.com/LaZsolt/delayMicroseconds/tree/master/for_MCUdude/function
have you actually tried some of the code on different clock speed to verify that is also does what it should in practice?

LaZsolt, 2020-09-10:
Yes, this simple inline assembly above will work all F_CPU values. ArduinoCore-samd delay.h defines almost the same code for delayMicroseconds().
I think may useable for microcore too. 
But it has many limitations, which mean **it is not fully compatibile** with the old Arduino-AVR or current MiniCore delayMicroseconds() function.
- The argument have to be compile time constants. In this case timing better than in current MiniCore.
- Not recommended to use with runtime variable argument. It cause huge overhead in timing.
- It always delays 1 more clock cycle on frequencies divisible by 4.
- Requested microsecond cannot be 0, but using of __builtin_constant_p() can correct this mistake.
- Requested microsecond cannot be 0, 1, 2, 3 on 1 MHz and cannot be 0, 1 on 2 MHz.

I tested it in my IOT web temperature sensor on 16 MHz, with DS18B20 sensors and ENC28J60 ehernet controller.
I have boards only on 16 MHz.
 3 years ago matthijskooijman created a delay tester program, later I will run to test my code.

Calculated cyles and delay in microseconds (looks a bit ugly) :

| Argument: | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 |  17 | 33 | 143 |
| :--: | :-----: | :-----: | :-----: | :-----: | :-----: | :-----: | :-----: | :-----: | :-----: | :-----: | :-----: | :-----: |
| 32 MHz | 33 - 1.03 | 65 - 2.03 | 97 - 3.03 | 129 - 4.03 | 161 - 5.03 | 193 - 6.03 | 225 - 7.03 | 257 - 8.03 | 289 - 9.03 | 545 - 17.03  | 1057 - 33.03 | 4577 - 143.03 |
| 20 MHz | 21 - 1.05 | 41 - 2.05 | 61 - 3.05 | 81 - 4.05 | 101 - 5.05 | 121 - 6.05 | 141 - 7.05 | 161 - 8.05 | 181 - 9.05 | 341 - 17.05 | 661 - 33.05 | 2861 - 143.05 |
| 16 MHz | 17 - 1.06 | 33 - 2.06 | 49 - 3.06 | 65 - 4.06 | 81 - 5.06 | 97 - 6.06 | 113 - 7.06 | 129 - 8.06 | 145 - 9.06 | 273 - 17.06 | 529 - 33.06 | 2289 - 143.06 |
| 12.5 MHz | 13 - 1.04 | 25 - 2 | 37 - 2.96 | 49 - 3.92 | 61 - 4.88 | 73 - 5.84 | 85 - 6.8 | 101 - 8.08 | 113 - 9.04 | 213 - 17.04 | 413 - 33.04 | 1785 - 142.8 |
| 11.0592 MHz | 9 - 0.81 | 21 - 1.9 | 33 - 2.98 | 45 - 4.07 | 53 - 4.79 | 65 - 5.88 | 77 - 6.96 | 89 - 8.04 | 97 - 8.77 | 189 - 17.09 | 365 - 33 | 1581 - 142.96 |
| 9.6 MHz | 9 - 0.94 | 17 - 1.77 | 29 - 3.02 | 37 - 3.85 | 49 - 5.1 | 57 - 5.94 | 65 - 6.77 | 77 - 8.02 | 85 - 8.85 | 161 - 16.77 | 317 - 33.02 | 1373 - 143.02 |
| 8 MHz | 9 - 1.13 | 17 - 2.13 | 25 - 3.13 | 33 - 4.13 | 41 - 5.13 | 49 - 6.13 | 57 - 7.13 | 65 - 8.13 | 73 - 9.13 | 137 - 17.13 | 265 - 33.125 | 1145 - 143.125 |
| 2 MHz | 262145 - 131072.5 | 5 - 2.5 | 5 - 2.5 | 9 - 4.5 | 9 - 4.5 | 13 - 6.5 | 13 - 6.5 | 17 - 8.5 | 17 - 8.5 | 33 - 16.5 | 65 - 32.5 | 285 - 142.5 |
| 1 MHz | 262145 - 262145 | 262145 - 262145 | 262145 - 262145 | 5 - 5 | 5 - 5 | 5 - 5 | 5 - 5 | 9 - 9 | 9 - 9  | 17 - 17 | 33 - 33 | 141 - 141 |

MCUdude, 2020-09-11:
For MiniCore, and all the other Arduino cores for ATmega's (Mighty/Mega/MajorCore), I think it's best to keep compatibility and use the function you provided instead of the macro. I bet there is a lot of code out there that depends on delay times that are not known at compile time.

The tables you posted above, is this data from the macro or from the function?

MCUdude, 2020-09-11:
I connected my trusty signal generator to the clock input of the AVR development board I'm using to get a very accurate clock. I made a simple sketch that toggles a pin by witing to PINB. I tested your delayMicroseconds code with all clocks MiniCore supports, and it looks very good. I'm going to merge this code into [MCUdude_corefiles](https://github.com/MCUdude/MCUdude_corefiles). This way the new delayMicroseconds function will be available on MightyCore, MiniCore, MegaCore, and MajorCore. Thanks for your contribution!

MCUdude, 2020-09-11:
The updated delayMicroseconds function will be available in the next MiniCore release!

closed 2020-09-11 by MCUdude