MCUdude/MiniCore #131: Problem compiling while using the wire library ( Wire.onRequest )

protttoy, 2020-03-18:
The code compile while using the regular arduino boards (UNO, Mini Pro) , but when using the minicore boards it does not compile and says this 

> invalid conversion from 'void (*)()' to 'void (*)(int)' [-fpermissive]

when calling **Wire.onReceive(i2c_when_receive);** the program does not compile
and the **i2c_when_receive function** is:

```
void i2c_when_receive()
{
  while (0 < Wire.available()){
     i2c_command = Wire.read();
  }
  
  #ifdef  DEBUG
    Serial.print("recived: ");
    Serial.print(i2c_command);
    Serial.println();
  #endif
}
```

protttoy, 2020-03-18:
it works when I use this
```
void i2c_when_receive(int x)
{
  while (0 < Wire.available()){
     i2c_command = Wire.read();
  }
  
  #ifdef  DEBUG
    Serial.print("recived: ");
    Serial.print(i2c_command);
    Serial.println();
  #endif
}
```

harryberlin, 2020-03-18:
my solution is, compiling and exporting hex with usual boards. after this, i do upload the hex with avrdude.
because the compiled program with usual boards is smaller.

MCUdude, 2020-04-13:
It does not compile because MiniCore is (on purpose) a little more picky on what code is allowed. You can have a read about the -fpermissive flag. MiniCore won't compile because your code has an issue. The -fpermissive flag (that the Uno and Nano has set, but MiniCore does not) mutes this warning, and that's a bad thing.

closed 2020-04-13 by MCUdude