MCUdude/MiniCore #132: Unify SPI / SPI1 Classes

vpreatoni, 2020-03-26:
Hi,
I'm working on an incompatibility issue related to Greiman's SdFat lib and MiniCore ATMEGA328PB dual SPI uC.

MiniCore defines two different classes for SPI ports: SPI.h for first SPI, and SPI1.h for sencond SPI. This is a bit problematic when trying to use SdFat lib (SD card connected to SPI1), since SdFat will only accept an SPIClass * paramenter, and not SPIClass1 *. Tryed casting but it doesn't work.

I've noted that SAMD board packages [https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SPI](url)  defines a single SPI class, but will accept &PERIPH_SPI as a constructor to define which SPI port will be used.
```
SPIClass SPI (&PERIPH_SPI,  PIN_SPI_MISO,  PIN_SPI_SCK,  PIN_SPI_MOSI,  PAD_SPI_TX,  PAD_SPI_RX);
SPIClass SPI1(&PERIPH_SPI1, PIN_SPI1_MISO, PIN_SPI1_SCK, PIN_SPI1_MOSI, PAD_SPI1_TX, PAD_SPI1_RX);
// more instances follow
```

Is it possible to fix this and use SAMD standard?

PS: this is the link to SdFat issue report: [https://github.com/greiman/SdFat/issues/180](url)

MCUdude, 2020-03-26:
It's not that easy on classical AVRs like the 328PB. On the SAMD every sercom has its own object, so it's much more portable. Classic AVRs are different. Basically, the entire SPI and SPI1 libraries have to be rewritten, and all registers has to be passed as arguments to the SPIClass constructor I believe.

vpreatoni, 2020-03-26:
Ops... okay. Will find another way.

BTW, any recommendation on how to access SD card on SPI1 port?. Arduino SD libraries are hardcoded to SPI.h. My last hope was Greiman's SdFat lib, but  neither works.

MCUdude, 2020-03-26:
You can always for the [official SD library](https://github.com/arduino-libraries/SD/) and modify the code to use SPI1/SPIClass1. It should be a straight forward job since the SD library only calls the SPI library, not the underlying registers (which SPI.h/SPI1.h takes care of).

I'm closing this issue because it's not really MiniCore's fault that classic AVR registers aren't portable like it is on the SAMD (and the new megaAVR-0 series). But even though it's closed I can still help you out here if you have more questions 🙂

closed 2020-03-26 by MCUdude