MCUdude/MiniCore #133: 8MHz External programming baud rate increase can cause issues for existing projects

easytarget, 2020-04-05:
This is more of a 'FYI' than an request for a 'fix' since a relatively good workaround is available. Commit ef51c5129a0523482fcee8745add26b74a9b2ccf caused me issues when I came to update an existing 328P based project.

I use the Arduino IDE and found myself suddenly unable to upload new code after updating to v2.0.4 via the boards manager. This was somewhat confusing since everything else worked; serial console was Ok, the CPU was resetting properly, but then I got the classic avrdude *'please wait while I timeout 10 times'* failure. A quick hunt led me to the discussions in #115.

The installed 8MHz External bootloader was programmed last summer, and so expects 38400 baud for programming, burning a new bootloader on this assembled and installed project is not trivial. 

Fortunately selecting `8MHz internal` in the clock list proved to be a quick fix since this clock setting still uses 38400 baud for the upload. This is a good workaround for anyone who is unwilling or unable to update their bootloader.

Looking for a better solution for myself I tried the following in `boards.txt` and it works as expected; at the expense of yet another entry in the clock menu:
```
--- boards.txt	2020-04-06 00:43:10.075499147 +0200
+++ boards.txt.orig	2020-04-06 00:59:08.427431968 +0200
@@ -147,11 +147,6 @@
 328.menu.clock.8MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
 328.menu.clock.8MHz_external.build.f_cpu=8000000L
 
-328.menu.clock.8MHz_external_slow=External 8 MHz - slow serial
-328.menu.clock.8MHz_external_slow.upload.speed=38400
-328.menu.clock.8MHz_external_slow.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
-328.menu.clock.8MHz_external_slow.build.f_cpu=8000000L
-
 328.menu.clock.7_3728MHz_external=External 7.3728 MHz
 328.menu.clock.7_3728MHz_external.upload.speed=115200
 328.menu.clock.7_3728MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
```
I've got a feeling others may get bitten by this, so something similar could be applied to all the relevant entries in `boards.txt`; or  this can just be documented/noted and handled by advising the selection of '8MHz internal' until the target bootloader is can be re-programmed.

If you agree with adding a menu entry and would like a PR just let me know and I'll do it sometime when it' not 2am, and I haven't been fighting arduino issues all day. ;-)

MCUdude, 2020-04-06:
I've been thinking a lot about this, and I do understand that this causes breakage for many users. More than I initially thought. I think it may be best to revert this change and do a 2.0.5 release very soon? TBH the absolute best solution was if it ran on 57600 from the beginning. However, 38400 has been the standard MiniCore baud rate for 8 MHz external for years and it may be too late to change this now.

I'm not a huge fan of too many submenus, so it's better if it "just works" like it used to do.

easytarget, 2020-04-06:
Having released 2.0.4 there will be quite a few users who have now burned with a 57600 bootloader, so reverting this will impact them;  I suspect you are damned whichever way you go, and without some sort of 'speed detect' feature in avrdude there is nojust works solution possible.

The best way forward is probably to accept the extra menu entries, the clock speed list is already quite long so this not going to affect the balance too much. Both pre- and post- 2.0.4 users will be catered for, and  you can keep 57600 as the default going forward, which is good. Plus you can stop worrying and move on ;-)

ZOUDIO, 2021-05-18:
I have a lot of PCBs in the field using an older Minicore version. Is there a possible modification to the latest version of the bootloader without breaking compatibility with older bootloaders. Then I can ship any new boards with the latest bootloader

MCUdude, 2021-05-19:
> I have a lot of PCBs in the field using an older Minicore version. Is there a possible modification to the latest version of the bootloader without breaking compatibility with older bootloaders. Then I can ship any new boards with the latest bootloader

Are these boards running on an external 8 MHz clock? Like mentioned in this thread, the baud rate for 8 MHz external clock was changed from 38400 to 57600. If you have devices in the field you can either use MiniCore v2.0.3 or you can change baudrate in boards.txt quite easily:

https://github.com/MCUdude/MiniCore/blob/07c2c096daabbf5c8045bc284472fd0a76056a7a/avr/boards.txt#L171-L176

You can also set your preferred upload baudrate in a PlatformIO project.

MCUdude, 2021-12-19:
Seems like the dust has settled. I'm closing this issue.

To anyone that may stumble across this issue: **Re-burn the bootloader, and it should be all good!**

closed 2021-12-19 by MCUdude