MCUdude/MiniCore #136: avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

lwh456789, 2020-04-20:
I'm using Mega 2560, so I can compile and upload successfully, but I can't burn it, so what's the problem
[image]
[image]
[image]

MCUdude, 2020-04-20:
You will need a dedicated programmer if you want to burn the bootloader. The first screenshots show us that Avrdude can't find the programmer you've selected, which is the AVRISPmkII. Do you have an AVRISPmkII?

lwh456789, 2020-04-20:
It was AVRISP MKII
[image]

lwh456789, 2020-04-20:
[image]

MCUdude, 2020-04-20:
Yes, but do you actually have that programmer connected to your PC? If yes, download [Zadig](https://zadig.akeo.ie) and choose one of the libusb drivers to use for the AVRISPmkII.

lwh456789, 2020-04-20:
yes

lwh456789, 2020-04-20:
how to use Zadig

lwh456789, 2020-04-20:
[image]

MCUdude, 2020-04-20:
You will have to click Options > List all devices. The select AVRISPmkII and select one of the libusb drivers

lwh456789, 2020-04-20:
[image]

lwh456789, 2020-04-20:
I'm sorry. Could you be a little more specific

lwh456789, 2020-04-20:
I can't find AVRISP MKII

MCUdude, 2020-04-20:
If you have an AVRISPmkII connected it will shop up in the drop-down menu where currently **Integrated camera** is selected.

lwh456789, 2020-04-20:
[image]

lwh456789, 2020-04-20:
but I cann't find it

lwh456789, 2020-04-20:
what's the problem

lwh456789, 2020-04-20:
The serial port that connects to my PC IS USB 2.0  serial

MCUdude, 2020-04-20:
Well, you computer can't find your programmer. I can't help you with that. You can always have a look in the windows device manager to see if it shows up.

Just to make sure, you have THIS DEVICE connected, right?

[image]

lwh456789, 2020-04-20:
Like the Avr device you showed in the picture？I looked around my computer and didn't see the device you showed the picture on. Do I need to install this device for programming with AVRISPM KII？

MCUdude, 2020-04-20:
OK, what do you want to do?

If you want to upload using the standard USB connector on your board, you just select your serial port and hit upload.

If you want to change other settings, like BOD or clock frequency, you will have to burn the bootloader. And for that, you **need a dedicated programmer** just like the one in the picture. If you don't have one, you cannot burn the bootloader.

If you absolutely have to do this, I recommend you get a USBasp programmer.

lwh456789, 2020-04-20:
Yeah, I'M GONNA burn the bootloader to my MEGA 2560

lwh456789, 2020-04-20:
Maybe I need to buy an AVRISPM KII right now?

lwh456789, 2020-04-20:
In that case, after purchasing the AVRISPM KII device, I'll try burning the bootloader again，Thank you for your guidance

MCUdude, 2020-04-20:
Yes do that. Or a USBasp programmer.

closed 2020-04-20 by MCUdude