MCUdude/MiniCore #140: Unable to upload Sketches ATmega328@1 MHz

marcmoe, 2020-05-26:
After upgrading MiniCore from v2.0.4 to v2.0.5 I am no longer able to upload sketches when using the 1 MHz bootloader with the internal oscillator on ATmega328P. I am able to burn the bootloader using the Arduino IDE and Arduino as ISP without issues but I can't upload any Sketeches using my FTDI.

`Arduino: 1.8.12 (Windows 10), Board: "ATmega328, Yes (UART0), 328P / 328PA, BOD 1.8V, LTO disabled, Internal 1 MHz"

Der Sketch verwendet 1054 Bytes (3%) des Programmspeicherplatzes. Das Maximum sind 32256 Bytes.
Globale Variablen verwenden 9 Bytes (0%) des dynamischen Speichers, 2039 Bytes für lokale Variablen verbleiben. Das Maximum sind 2048 Bytes.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\kleig\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf -v -patmega328p -carduino -PCOM12 -b9600 -D -Uflash:w:C:\Users\kleig\AppData\Local\Temp\arduino_build_940283/Blink_solid.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\kleig\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : COM12
         Using Programmer              : arduino
         Overriding Baud Rate          : 9600
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xe8
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xe8

avrdude done.  Thank you.

Problem beim Hochladen auf das Board. Hilfestellung dazu unter http://www.arduino.cc/en/Guide/Troubleshooting#upload.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`
I've tried two on differnt ATmega328P and two different FTDIs without success.
When selecting other booloader options like 2 MHz  or 4 MHz with the internal oscillator or the 1 MHz bootloader from MiniCore v2.0.4 I am able to upload sketeches.

MCUdude, 2020-05-26:
Thank you for reporting! It means a lot 🙂

There was a tiny change in the bootloader source code from 2.0.4 to 2.0.5, so I decided to try this on some actual hardware. The first thing that his me is that the bootloader LED is flashing way too fast. According to my oscilloscope, it's flashing 8 times faster than it should! This means that the CPU is also running at 8MHz and not 1 MHz as it should. This means you would probably be able to upload with a baud rate of 9600*8 = 76800.

It turns out that the [tiny change I did to support 4 and 2 MHz using the internal clock prescaler](https://github.com/MCUdude/optiboot_flash/blob/1a10e0cecbe120fc7e6369cc2292c60cc82b1d92/optiboot_flash.c#L517-L521) has backfired. Instead, it should say:

```c
#if defined(CLKPR) && F_CPU != 1000000L
  CLKPR = 0x80; // Enable the clock prescaler
  CLKPR = 0x00; // Set prescaler to 1
#endif
```

I will push a fix for this soon, and it affects all my Arduino cores, not just this core and ATmega328P.

MCUdude, 2020-05-27:
I've just pushed a fix, and it works like it should on my hardware.

Please do a manual install of MiniCore and see if it works for you. I'll probably have to release 2.0.6 soon because of this 🙂

closed 2020-05-27 by MCUdude

marcmoe, 2020-05-27:
Thank you for your quick response. Uploading sketches works again. I've tested it by installing the master branch from zip download.

I'm looking forward for the next release 😃 Great work

MCUdude, 2020-05-27:
Thank you for reporting!

It is comprehensive bug reports like this that's fun to work with. All relevant information is there, and it's all about finding the bugs and fixing them.

lrusnac, 2020-08-30:
hey @MCUdude, could you push a new release please? or at least reopen the issue? I did spent a couple of hours debugging this exact problem today, and it's been a while since you fixed this.
after manually installed the master branch it all worked out

MCUdude, 2020-08-30:
@lrusnac Thanks for notifying me, I totally forgot to push a new release. v2.0.6 is now available through boards manager!

commit 455f48d1 (2020-05-27): Update Optiboot_flash
Fix an issue where bootloaders compiled for 1 MHz was actually running at 8 MHz.
Closes #140
Merge commit '1e929712ea9d85cd8726b30fa8cab7c7bdaaa620'