MCUdude/MiniCore #141: cannot choose atmega8 from board manager

martinbandung, 2020-05-28:
hi, i have installed the board minicore using board manager. I want to use atmega8. But the option atmega8 is grayed out, it canoot be chosen. only atmega8 cannot be chosen. I'm using ubuntu 20.04.

here the screen shoot
https://drive.google.com/file/d/11TdySDuOOv7SCDrFJpmLzHlnAyK7UUVO/view?usp=sharing

my system info

id: | cpu
-- | --
description: | CPU
product: | Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz
vendor: | Intel Corp.
physical id: | 11
bus info: | cpu@0
version: | Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz
serial: | NULL
slot: | SOCKET 0
size: | 1497MHz
capacity: | 2700MHz
width: | 64 bits
clock: | 100MHz
capabilities: | lm fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp x86-64 constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap intel_pt xsaveopt dtherm ida arat pln pts md_clear flush_l1d cpufreq
configuration: | cores=2enabledcores=2threads=4 | cores | = | 2 | enabledcores | = | 2 | threads | = | 4
cores | = | 2
enabledcores | = | 2
threads | = | 4

MCUdude, 2020-05-28:
Your screenshot shows that **MiniCore** is greyed out (which is the name), but ATmega8 is not.

martinbandung, 2020-05-29:
ah, i think i knew the problm, the list goes through over the dock, so it cannot be selected

closed 2020-05-29 by martinbandung