MCUdude/MiniCore #142: Serial output not working

dpetican, 2020-05-28:
I'm able to program a 328P from Arduino v1.8.10 using an Atmel ICE (AVR) device.  MiniCore is v2.05 The oscillator I'm using is 8MHz internal and the voltage is 5.0V.  However, no matter what I do I can't seem to get any serial output.  PD0/PD1 are connected to TX/RX.  I've tried switching TX/RX.  Also, I cannot burn the bootloader.  I do have Atmel Studio installed and can upload a program using it. If its a matter of the fuses I'm not sure what to look for.  Thanks.

Arduino: 1.8.10 (Windows 10), Board: "ATmega328, Yes (UART0), 328P / 328PA, BOD  disabled, LTO enabled, Internal 8 MHz"

C:\Users\dpeti\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\dpeti\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf -v -patmega328p -catmelice_isp -Pusb -e -Ulock:w:0x3f:m -Uefuse:w:0b11111111:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xe2:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\dpeti\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : atmelice_isp
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : Atmel-ICE (ARM/AVR) in ISP mode
         Vtarget         : 4.9 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
[... 118 lines cut ...]
avrdude: 32768 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Error while burning bootloader.
Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x0f
avrdude: verification error; content mismatch

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MCUdude, 2020-05-28:
It seems like it has problems accepting the lock bit. Can you try to open your command lineand enter these bare commands instead?

```
C:\Users\dpeti\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\dpeti\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf -v -patmega328p -catmelice_isp -Pusb -e -Ulock:w:0x3f:m -Uefuse:w:0b11111111:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xe2:m
```

And then:

```
C:\Users\dpeti\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\dpeti\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/avrdude.conf -v -patmega328p -catmelice_isp -Pusb -Uflash:w:C:\Users\dpeti\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.5/bootloaders/optiboot_flash/bootloaders/atmega328p/8000000L/optiboot_flash_atmega328p_UART0_38400_8000000L_B5.hex:i
```

dpetican, 2020-05-28:
Both commands worked.  I now have serial output and can upload via serial.  Not that I want to do that.  What's the role of the bootloader when it comes to serial comms and/or oscillator frequency?  Or was burning the bootloader just required to set the fuses correctly?  Thanks.

MCUdude, 2020-05-28:
The issue was the lock bit. I should probably look into it to make sure it doesn't happen to others as well.

The only thing bootloader provides is a way to upload new programs. Optiboot, which MiniCore uses, lets you upload code with a standard USB to serial adapter. If you don't need this, you can select "No bootloader" in Arduino IDE and run "Burn bootloader" again to only set the fuses.

dpetican, 2020-05-28:
Okay but why does uploading a program using my programmer (ISP) wipe out the bootloader?  I thought I had to hold down the shift key for this happen.  Two binaries always seem to be generated.  One with and one without the bootloader.  Since the lock bit wasn't programmed correctly were the other fuses programmed correctly?  How does this explain the serial output not working?  Just trying to understand the process so I can more effectively troubleshoot in the future.  Thanks.

MCUdude, 2020-05-28:
> Okay but why does uploading a program using my programmer (ISP) wipe out the bootloader?

The IDE uploads the binary without the bootloader included. I'm not sure this behavior can be changed or not. It's a well-known quirk at least.

> Since the lock bit wasn't programmed correctly were the other fuses programmed correctly?

You can look in the Avrdude output log to verify this.

```
avrdude: reading input file "0b11111111"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.09s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0b11111111:
avrdude: load data efuse data from input file 0b11111111:
avrdude: input file 0b11111111 contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0xd6"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.09s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0xd6:
avrdude: load data hfuse data from input file 0xd6:
avrdude: input file 0xd6 contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xe2"
avrdude: writing lfuse (1 bytes):
```

> How does this explain the serial output not working?

You're saying that you can't print to the serial monitor in your application, but the bootloader works?

MCUdude, 2020-05-28:
>> Okay but why does uploading a program using my programmer (ISP) wipe out the bootloader?

> The IDE uploads the binary without the bootloader included. I'm not sure this behavior can be changed or not. It's a well-known quirk at least.

It turns out it is possible to change this behavior so that the bootloader isn't wiped. Well, technically it is, but instead, it will load the hex file with the bootloader included. I'll push a fix for this soon.

dpetican, 2020-05-28:
I started writing this before your last post.  I appreciate your help.

I ran a couple of experiments and somethings are a little clearer now.  Upload using programmer will upload whatever binary was compiled (ie with/without bootloader) As soon as I compiled without bootloader and uploaded via ISP I could still upload via serial.

I'm assuming in the Arduino IDE compiling with bootloader is meant to be used with the burn bootloader option and not upload via programmer.  Now I am confused again.  The two commands you gave me did not reference any compiled program I generated with or without bootloader.  So whats the point of the option to compile with bootloader?  I think you answered that in the previous post.

Now why did the serial not work when I first programmed the orginal chip?  Because the LOW. CKDIV8 bit is set in new 328P parts.  The effective speed is I assume 1MHz and I compiled for 8MHz.  Clearing this bit using the device programmer in Atmel Studio and uploading via ISP I get serial output right away without burning a bootloader or changing any other fuse bits.

Compiling for 1MHz and uploding to another chip with LOW.CKDIV8 set also goves me serial output.  Good lessons learned.  Thanks.  This will help with a similar problem I'm having with another core and an ATtiny1614.

MCUdude, 2020-05-28:
> Upload using programmer will upload whatever binary was compiled (ie with/without bootloader) As soon as I compiled without bootloader and uploaded via ISP I could still upload via serial.

The IDE always creates two hex files; with and without bootloader, regardless of what options you've selected in the Tools menu. Be default the IDE would use the one without bootloader for serial uploads (makes sense, the bootloader is already present) and for uploading using a programmer. The latter will wipe the bootloader because of the Avrdude -e flag. However, I've just pushed a commit where the programmer will choose the hex file with the bootloader included. If you choose "No bootloader", the two compiled hex files will be the same.

> I'm assuming in the Arduino IDE compiling with bootloader is meant to be used with the burn bootloader option and not upload via programmer

I believe this file is generated so you can easily export the compiled binary, with and without a bootloader.

Just note that the latest release (2.0.5) has a bug in the bootloader where the 1 MHz one is running 8 times faster because I messed up the CLKPR register. A new release will be available soon, but meanwhile, if you want/need to run it at 1 MHz with a bootloader, you'll have to do a manual install for now.

dpetican, 2020-05-28:
Thanks.  Noted about 1MHz.  But I got what I need for now with this chip running at 8MHz.

closed 2020-05-28 by dpetican