MCUdude/MiniCore #143: CKSEL Fuses for External Clock Signal !

alianu, 2020-05-29:
Hello,
Can the MiniCore be updated in order to burn the bootloader with the fuses CKSEL to 0000 (CKSEL3, CKSEL2, CKSEL1, CKSEL0) for external clock signal. I want to use an SiTIME MEMS oscillator SIT2024BM-S1-33E-16.000000E for one of my projects.
Now the fuses are burn for External Clock generated by External Oscilartors, but I would like to use an External Clock Signal.
Thank you, best regards, Bela.

MCUdude, 2020-05-29:
There's nothing wrong with using the fuses like they are and apply a clock pulse from a driven oscillator. Just connect your clock signal to the XTAL1 pin, and leave the XTAL2 pin open. The microcontroller is just fine with this, I've done it many times 🙂

alianu, 2020-05-29:
Hello. Thank you for fast replay. So I should chose in MiniCore the option External 16MHz and just connect the XTAL1. I will try it thank you very much for info. Best regards, Bela.

MCUdude, 2020-05-29:
> Hello. Thank you for fast replay. So I should choose in MiniCore the option External 16MHz and just connect the XTAL1. I will try it thank you very much for info. Best regards, Bela.

That's correct!

alianu, 2020-05-29:
I have one more question if I may. If I use the fuses as they are, not 0000, and I use only XTAL1 pin for clock signal, can I use PB7 port for something else ? I am kind of short in ports and I need just one more. Will be just super great if will work.

MCUdude, 2020-05-29:
> If I use the fuses as they are, not 0000, and I use only XTAL1 pin for clock signal, can I use PB7 port for something else ? I am kind of short in ports and I need just one more. Will be just super great if will work.

I don't think that will work. When it's configured to drive a crystal the XTAL2 pin will act as an output pin that "kicks" the crystal to keep it oscillating. If you're so short of IO that you need PB7 as well you'll have to either modify your boards.txt file to change the fuses, use PlatformIO and define your own fuses in the platformio.ini file.

You probably already know this, but you can use the analog pins like any other digital pin. Well, except for A6 and A7. These can only be used for analog readings.

alianu, 2020-05-29:
So I will need to edit the boards.txt and edit the CLKSEL fuses I think. I have only A6 and A7 available, I can read inputs values and define which is equal to 0 logic (LOW) and which for 1 logic (HIGH) but cannot use them as output as I would like :( . Thank you anyway for the explanations. Best regards, Bela.

alianu, 2020-05-29:
one more question if I may. I did calculate the Low Fuses to be as I wanted, 0xF0, but in the boards.txt would I change this fuses only for ATMega328 on the Nano section or on all section where is ATMega328 ? I don't know how MiniCore works, if will select the board from this boards.txt file. Can you help me a bit if I don't ask to much. I did never edit the boards.txt for fuses or something else.
I did attach here my boards.txt file.
Thank you, best regards, Bela.
[boards.txt](https://github.com/MCUdude/MiniCore/files/4704684/boards.txt)

MCUdude, 2020-05-30:
It depends on what bootloader option you choose. If you change line 62 and 76, it should be suffient.

https://github.com/MCUdude/MiniCore/blob/ff3f6d62425e71ae2823f6e47f6d3af983bc37f9/avr/boards.txt#L62-L76

closed 2020-05-30 by MCUdude