MCUdude/MiniCore #145: Changing permission for /usr/share/.../create_disassembler_listing.sh: Operation not permitted

scrouthtv, 2020-06-22:
```
Linking everything together...
/home/lenni/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -Wall -Os -Wl,--gc-sections -mmcu=atmega168 -o /tmp/arduino_build_682403/Blink168.ino.elf /tmp/arduino_build_682403/sketch/Blink168.ino.cpp.o /tmp/arduino_build_682403/../arduino_cache_88997/core/core_bb969c779e035b4297159cb365195dae.a -L/tmp/arduino_build_682403 -lm
/home/lenni/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_682403/Blink168.ino.elf /tmp/arduino_build_682403/Blink168.ino.eep
/home/lenni/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_682403/Blink168.ino.elf /tmp/arduino_build_682403/Blink168.ino.hex
chmod +x /usr/share/arduino/hardware/MiniCore/avr/scripts/create_disassembler_listing.sh
chmod: changing permissions of '/usr/share/arduino/hardware/MiniCore/avr/scripts/create_disassembler_listing.sh': Operation not permitted
exit status 1
Error compiling for board ATmega168.
```

I manually changed the permissions on this file using sudo but I still get the same error.

closed 2020-06-22 by scrouthtv

MCUdude, 2020-06-22:
So, you figured it out?

scrouthtv, 2020-06-24:
Yeah I installed it into the wrong folder - I tried manual installation because at that time I had no working internet connection, so I installed everything into /usr/share/arduino/hardware/MiniCore

Which is the wrong folder