MCUdude/MiniCore #148: cannot set fuses nor burn bootloader via platformio

zakaster, 2020-08-17:
hello, been using minicore for quite a while via Arduino IDE
its kind of annoying that every time i have to burn bootloader via Arduino IDE and then code in VScode
until i saw `platformio.md`  mentioned those are doable via platformio as well so i gave it a shot
i used the exactly same template config file, with this line uncommented and modified
upload_protocol = stk500v1
since i'm using `arduino as isp` to burn the bootloader, however both command
pio run --target fuses
pio run --target bootloader
is reporting the similar error:
*** Do not know how to make File target `fuses' (C:\Users\xxx\OneDrive\workspace\embedded\arduino\__test_minicore\fuses).  Stop.

here is a detailed error from burning the fuses:
PS C:\Users\xxx\OneDrive\workspace\embedded\arduino\__test_minicore> pio run --target fuses
Processing MiniCore (platform: atmelavr; framework: arduino; board: ATmega328P)
-----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega328P.html
PLATFORM: Atmel AVR 1.15.0 > ATmega328P/PA
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
PACKAGES:
 - framework-arduinoavr 4.1.2
 - tool-avrdude 1.60300.190424 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
*** Do not know how to make File target `fuses' (C:\Users\xxx\OneDrive\workspace\embedded\arduino\__test_minicore\fuses).  Stop.

any idea what's going on?

zakaster, 2020-08-30:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

; ENVIRONMENT SETTINGS
[env:MiniCore]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = ATmega328P
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L

; HARDWARE SETTINGS
; Oscillator option
board_hardware.oscillator = external
; Hardware UART for serial upload
board_hardware.uart = uart0
; Brown-out detection
board_hardware.bod = 2.7v
; EEPROM retain
board_hardware.eesave = yes

; UPLOAD SETTINGS
board_upload.speed = 115200
; Upload serial port is automatically detected by default. Override by uncommenting the line below
;upload_port = /dev/cu.usbserial*

; BUILD OPTIONS
; Comment out to enable LTO (this line unflags it)
; LTO = link time optimization
build_unflags = -flto
; Extra build flags
build_flags = 

; Upload using programmer
upload_protocol = stk500v1
; Aditional upload flags
;upload_flags = -Pusb

; SERIAL MONITOR OPTIONS
; Monitor and upload port is the same by default
;monitor_port = 
; Serial monitor baud rate
monitor_speed = 9600

board_bootloader.lfuse = 0xFF
board_bootloader.hfuse = 0xDE
board_bootloader.efuse = 0xFD
board_bootloader.lock_bits = 0x0F
board_bootloader.unlock_bits = 0x3F
```

MCUdude, 2020-09-10:
Here's the output from my machine with the exact same platformio.ini contents as you:

```
Hanss-MacBook-Pro-2:atmega8535_test hans$ pio run -t fuses
Processing MiniCore (platform: atmelavr; framework: arduino; board: ATmega328P)
-----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega328P.html
PLATFORM: Atmel AVR (2.2.0) > ATmega328P/PA
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr-minicore 2.0.5 
 - tool-avrdude 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode

TARGET CONFIGURATION:
---------------------
Target = atmega328p
Clock speed = 16000000L
Oscillator = external
BOD level = 2.7v
Save EEPROM = yes
UART port = uart0
Clock output = no
---------------------

Selected fuses: [lfuse = 0xf7, hfuse = 0xd6, efuse = 0xfd]
```

The difference that I can see is that your Atmel AVR package is version 1.15.0 while mine is 2.2.0. Open PIO home and update your packages. Or run `pio update`

MCUdude, 2020-09-24:
Update PlatformIO and you're all good!

closed 2020-09-24 by MCUdude