MCUdude/MiniCore #149: Could you define Second i2c bus with using TwoWire class?

asukiaaa, 2020-08-20:
Some libraries (ex: [Adafruit_BME280](https://github.com/adafruit/Adafruit_BME280_Library/blob/master/Adafruit_BME280.h#L218), [SparkFun_ICM-20948_ArduinoLibrary](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/ICM_20948.h#L138)) support using second i2c bus but this platform use different class([TwoWire1](https://github.com/MCUdude/MiniCore/blob/e27fbe6733803d1ae37ddd67260e72044e7e77f3/avr/libraries/Wire1/src/Wire1.h#L34)) for second i2c bus so I cannot assign it to the libraries.(It causes type error.)
To solve the problem, could you define second i2c bus with using TwoWire class not with TwoWre1?
(Serial1 and SPI1 also have same problem.)

Thank you for sharing useful platform.
I use this platform for ATmega328PB on PlatformIO.

MCUdude, 2020-08-20:
I do understand that this will cause some issues, but it is a lot of work to rewrite the entire SPI and Wire library to fix this. BTW Serial1 does not have the same problems. All serial ports use `HardwareSerial` as their class.

It's not something I'll prioritize doing at the moment, but I'll keep it in mind.

MCUdude, 2020-08-20:
However, If you're capable of doing it, feel free to submit a PR!

mhmayyan, 2020-08-20:
@asukiaaa It might be easier for you to modify the libraries (Adafruit_BME280, SparkFun_ICM-20948_ArduinoLibrary). You need, for example, to replace "TwoWire" with "TwoWire1", and "#include <Wire.h>" with "#include <Wire1.h>".

asukiaaa, 2020-08-20:
Thank you for understanding.
Sorry, I didn't check Serial1. It is already no problem.

OK, if I try to update code about Wire on this weekend.

MCUdude, 2020-09-24:
Closed in 0cad0c6

closed 2020-09-24 by MCUdude

asukiaaa, 2020-10-01:
@MCUdude 
Could you release new version that includes the merging.
> Closed in 0cad0c6

MCUdude, 2020-10-04:
@asukiaaa there is still a few minor things I'm working on. Expect a new release in a few days!

asukiaaa, 2020-10-05:
I see, I wait releasing.

MCUdude, 2020-10-10:
Release 2.0.7 is now available!

asukiaaa, 2020-10-10:
Thank you for releasing :)

adoconnection, 2021-04-17:
So, should this snippet work if I connect SDA0 and SDA1, SCL0 and SCL1?
(it does not for me)

```// Send from Wire1 to Wire

#include <Wire.h>
#include <Wire1.h>

void setup()
{
  Wire1.begin(); // join i2c bus (address optional for master)
  Wire.begin(4);
  Wire.onReceive(receiveEvent);
  Serial.begin(9600);
  Serial.println("Wire1=master, Wire=slave");
}

byte x = 0;

void loop()
{
  Wire1.beginTransmission(4); // transmit to device #4
  Serial.println("1");
  Wire1.write("x is ");        // sends five bytes
  Wire1.write(x);              // sends one byte  
  Serial.println("2");
  Wire1.endTransmission();    // stop transmitting
  Serial.println("3");

  x++;
  delay(500);
}

void receiveEvent(int)
{
  while(1 < Wire.available())
  {
    char c = Wire.read();
    Serial.print(c);
  }
  int x = Wire.read();
  Serial.println(x);
}
```