MCUdude/MiniCore #154: atmega8 programming sketches via arduino ISP - wrong baudrate

DrDaveone, 2020-10-11:
Hey there! here's the thing: if i choose 8MHz, BOD 2.7, EEPROM ret, LTO disa, !!!BOOTLOADER Yes (UART0)!!!, Programmer: Arduino as ISP, sketch doesn't load due to baudrate = 38400 instead of 19200.
but if I change BOOTLOADER to NO, baudrate become 19200 and sketch loads correctly.

MCUdude, 2020-10-11:
>  if i choose 8MHz, BOD 2.7, EEPROM ret, LTO disa, !!!BOOTLOADER Yes (UART0)!!!, Programmer: Arduino as ISP, sketch doesn't load due to baudrate = 38400 instead of 19200.

This is because the target chip you're trying to program doesn't have a bootloader, or has the wrong one. If you want to use a bootloader to upload code, you'll have to burn the bootloader first. The bootloader baud rate is 38400, while the Arduino as ISP baud rate is 19200. If the fusebits are already set, you can hold the shift key while clicking upload to upload using the programmer.

> but if I change BOOTLOADER to NO, baudrate become 19200 and sketch loads correctly.

If you don't want to use a bootloader or don't have a USB to serial adapter, use this option.

DrDaveone, 2020-10-11:
Alright, i get it
handled with bootloader & serial connection, it works, thanks!

closed 2020-10-11 by DrDaveone