MCUdude/MiniCore #155: Can't upload via Arduino IDE onto a ATmega 328P with the internal oscillator

scrouthtv, 2020-10-18:
Hi,
for a project I'm trying to get this ATmega running without an external oscillator. My project is not time-critical, so I guess I could save some money on the quartz.
Now when I create a very basic circuit with the ATmega *and an external 16 MHz oscillator* and these settings:
[image]
Uploading works fine.
However, when I remove the oscillator and set the `Clock` option to either `Internal 8 MHz`, `Internal 4 MHz`, `Internal 2 MHz` or `Internal 1 MHz`, I get this error:
```
Sketch uses 1462 bytes (4%) of program storage space. Maximum is 32256 bytes.
Global variables use 11 bytes (0%) of dynamic memory, leaving 2037 bytes for local variables. Maximum is 2048 bytes.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
```

In the README, I read to change the baud rate to something lower, so I edited boards.txt#221 to this:
```
328.menu.clock.8MHz_internal.upload.speed=19200
```
And relaunched the IDE. Uploading fails with the same error. Should I try the other internal clocks? With which baud rates?

Do I have to calibrate the internal oscillator? Is there any way to use an external clock for uploading but the internal clock at runtime?

Thanks for any help in advance.

scrouthtv, 2020-10-18:
Oh also, I think I'm running v2.0.7

MCUdude, 2020-10-18:
There are several settings you'll have to "Burn Bootloader" in order to load the new settings into the microcontroller. For instance; EEPROM retain, BOD, and clock option. Select the settings you want, and burn the bootloader. The correct bootloader hex file and the correct fuse bits will be loaded, and it will work as expected. You'll need a dedicated programmer to do this.

scrouthtv, 2020-10-26:
Do I need to burn the bootloader everytime I change the settings? Because I already installed the bootloader using another Arduino and this guide: https://www.arduino.cc/en/Tutorial/BuiltInExamples/ArduinoToBreadboard

MCUdude, 2020-10-26:
No, you only need to burn the bootloader once. Does it work using the bootloader if you keep the external oscillator, and use 16 MHz? It may be because the internal oscillator isn't very accurate.

scrouthtv, 2020-10-26:
Yes, it does work with the oscillator.
If I remove it and change the property in the IDE, I get this error.
I also tried with a lower baudrate

MCUdude, 2020-10-26:
OK. I'll wire up a board exactly as you describe it and see if I run into the same problem. If I don't, it is because the internal oscillator in your chip is probably a bit too slow. I'll keep you updated.

scrouthtv, 2020-10-26:
This is what I did in the first step:
 
[image]

Uploading works fine.

When I remove XT I get the error above.

Thanks for your help in advance

MCUdude, 2020-10-26:
I did test this on an ATmega328P, and everything worked fine.

* Install the 16 MHz oscillator and burn the bootloader. Verify that everything is working by uploading to the microcontroller using the USB to serial adapter
* Select these settings
  [image]
* Burn the bootloader again, with the 16 MHz crystal installed
* Remove the crystal. If you connect an LED to digital pin 13, it should flash twice if you reset the microcontroller
* Try uploading using the USB to serial adapter

SpenceKonde, 2020-10-27:
Any time you change the Clock, BOD, EEPROM, or Bootloader options, you must burn bootloader again in order to set the fuses correctly (with very few exceptions- changing between two crystals of different speed, but within the same "speed bracket" in terms of the CSEL bits of lfuse, or between two different speeds derived from the internal 8 MHz oscillator neither of which are 1 MHz, unless it's a tiny4313 or tiny2313, which has two internal oscillators... and if that made your head spin, well, that's why I tell people to always burn bootloader if they change any of those options on any classic* AVR).  

* classic being anything released before 2016; the modern AVRs have a radically different architecture; the parts supported by megaTinyCore and MegaCoreX don't need you to burn bootloader when changing clock speed unless you're changing to or from a speed divisible by 5 - and the latest and greatest in the land of AVRs, the Dx-series, supported by DxCore, never need you to butrn bootloader to change clock speed/source... You see, it's all very simple and straightforward! Clear as crystal! >.>

scrouthtv, 2020-10-27:
Ok, I will try that as soon as I'm back home. Thanks for the help!

MCUdude, 2020-11-20:
So, how did it go?

ajinkya-open, 2021-02-05:
I witnessed the same not able to upload by internal acillator, if any one guide how to check it, i would like to contribute in the this project,
so if any one has idea, inform me i can check any modifications

cburstedde, 2021-02-24:
> [image]

This may not answer any question, but depending on what the XT device is, it might require additional capacitors from each TOSCX to ground. The usual crystals for example work with two 18pF or 22pF caps. And, at the risk of being trivial, a 100nF capacitor between VCC and ground is also required.

pfeerick, 2021-03-15:
I was just flashing some 16Mhz EXT OSC ATMega328s to 1Mhz INT OSC yesterday, hence did the burn bootloader because of changed fuse settings. It appears I also need to use the low speed USBASP option when programming via ISP otherwise the target will not be detected. I didn't need to change any `boards.txt` settings to upload via serial - flashing the bootloader, which also sets the correct clock fuses, is all that was needed when changing speeds.

cimba007, 2021-04-02:
Got the same problem .. with internal OSC I am not able to upload any sketch. This somehow lead me back to some posts from tha past ... https://github.com/Optiboot/optiboot/issues/209 still not shure if it is related.

duraluminium, 2021-06-05:
I experienced a similar error, I know the thread is old but maybe this will help others landing here. After burning bootloader, in order to upload a sketch you have to either Shift+Upload or select **"Upload with programmer"** from the menu bar! Clicking "upload" will try and upload to the programmer/arduino, NOT the 328 chip being targeted. The error results from wrong parameters the arduino doesn't accept.

MCUdude, 2021-06-06:
I'm closing this issue because OP hasn't replied in a long time, and it's most likely a hardware issue.

The issue @pfeerick pointed out where a slow programmer option had to be used is a known "issue" with the USBasp programmer. Its default speed is too fast for an AVR running at 1 MHz.

@duraluminium's fix doesn't really solve the issue here, because he's uploading using a programmer rather than trying to get the bootloader to work.

closed 2021-06-06 by MCUdude

scrouthtv, 2021-06-06:
> OP hasn't replied in a long time

Didn't find the time for this as I started university. Might get back to this at some time later, but closing is fine with me.