MCUdude/MiniCore #156: system cannot find the file specified...

ZL1CVD, 2020-10-27:
Something wrong with latest boards.txt (I think). Just upgraded to latest version via boards manager & compiler is not throwing up a file error when I export HEX :
Sketch uses 12822 bytes (39%) of program storage space. Maximum is 32768 bytes.
Global variables use 261 bytes (12%) of dynamic memory, leaving 1787 bytes for local variables. Maximum is 2048 bytes.
C:\Users\Chris\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.7/scripts/delete_merged_output.bat false C:\Users\Chris\AppData\Local\Temp\arduino_build_922559\CIV-SMeter.ino.with_bootloader.hex 
cmd /C copy C:\Users\Chris\AppData\Local\Temp\arduino_build_922559\CIV-SMeter.ino_atmega328pb_16000000L.lst E:\PRODUCTS\CAT Icom\CIV-SMeter 
The system cannot find the file specified.
Error compiling.

I'm compiling 328PB with LTO enabled, no EEPROM retailed, no bootloader and exporting HEX via Sketch >Export Compiled Binary.

per1234, 2020-10-27:
I believe you have encountered this issue:
https://github.com/arduino/arduino-cli/issues/958#issuecomment-701149126

ZL1CVD, 2020-10-27:
Thanks for the response but no, I removed space in path and still have error. Anyway thinking about it, it was working fine, with space in folder name, until I updated to latest version of Minicore..

MCUdude, 2020-10-27:
A batch script (delete_merged_output.bat) was introduced in the latest version of MiniCore (v2.0.7), and this is what's causing the issue. The script is there so it's easier to look through the assembly output of your program if you wish to do so.

I'm not sure if @SpenceKonde has seen this before or not. After all, it's his script. If there isn't a proper fix for this I'll probably remove the assembly output option for Windows until this is sorted out.

SpenceKonde, 2020-10-27:
I removed delete_merged_output.bat from megaTinyCore quite a while ago, after I discovered that the IDE wasn't generating merged output files at all for boards where a bootloader wasn't in use, and there was no need for it. But MiniCore does seem to be generating them (so does ATTinyCore - I haven't dug into why yet). 

But the problem reproduces for me too. It looks to me like the problem is that you're trying to do:

```
cmd /C copy C:\Users\Spence\AppData\Local\Temp\arduino_build_904133\BlinkCopy.ino_atmega328p_16000000L.lst C:\Users\Spence\Documents\Arduino\sketches\BlinkCopy 
```

But the file present isn't called BlinkCopy.ino_atmega328p_16000000L.lst , it's called BlinkCopy.ino.lst

MCUdude, 2020-10-27:
@SpenceKonde thanks for the clue! turns out the .lst file is incorrectly named when compiling on a Windows computer.

https://github.com/MCUdude/MiniCore/blob/master/avr/platform.txt#L78

@ZL1CVD try to replace

` "{build.path}/{build.project_name}.lst"`
with
`"{build.path}/{build.project_name}_{build.mcu}_{build.f_cpu}.lst"`

if so, I'll push the fix very soon

SpenceKonde, 2020-10-27:
Yup, that fixes it!

And yes, I have a sketch called BlinkCopy - an exact copy of blink, saved outside of examples just so I can export compiled binary (lol - Most recently it was relevant for fixing the names of my exported .hex and .lst files; I now wrap up every tools submenu option that effects the compiled binary, ex: BlinkCopy.ino.t1614.20c0.u5V.mD0.hex (ATtiny1614 at 20 MHz clocked from internal, with 5V baud rate correction and millis clocked from TCD0), but for a while kept having stupid situations where the name came out mangled, though I've definitely used it for looking at how the millis ISR was coming out pretty frequently.

SpenceKonde, 2020-10-27:
Oh - and you need to update version in platform.txt

ZL1CVD, 2020-10-27:
Hi all.
Just confirming changing 

"{build.path}/{build.project_name}.lst"

with

"{build.path}/{build.project_name}_{build.mcu}_{build.f_cpu}.lst"

in file plaform.txt

does indeed fix the problem.

Huge thanks to all!!

commit 96f4a0a8 (2020-10-27): Fix assembly listing file path for Windows
Closes #156
closed 2020-10-27 by MCUdude