MCUdude/MiniCore #157: Clock prescaler and MiniCore

strange-v, 2020-11-07:
Currently, I use ATmega328P with a 4MHz external oscillator to be able to run MCU in the voltage range from 1.8V to 3.6V. The main disadvantage of 4MHz crystal is size, I'm not able to find SMD 3225 or 5032 locally.

I've just remembered that this MCU has a clock prescaler, setting it to 4 I should get the same 4 MHz frequency (and ability to start at 1.8V) with a 16 MHz oscillator.

Can I easily achieve this with MiniCore? Theoretically, I should modify "External 4 MHz" to set CLKPS bits during the burn bootloader process.
```
328.menu.clock.4MHz_external=External 4 MHz
328.menu.clock.4MHz_external.upload.speed=9600
328.menu.clock.4MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
328.menu.clock.2MHz_internal.build.clkpr=-DOSC_PRESCALER=0x02
328.menu.clock.4MHz_external.build.clock_speed={build.f_cpu}
328.menu.clock.4MHz_external.build.f_cpu=4000000L
```

Is this correct?

PS
More about the project on [hackaday.io](https://hackaday.io/project/175514-wireless-solar-powered-sensor-node).

MCUdude, 2020-11-08:
Hi!

I would suggest you just add this part to the boards.txt file. It would probably not work with the bootloader since it's not compiled to support a 16 MHz crystal being divided by four.

```
328.menu.clock.16_4MHz_external=External 16/4 MHz
328.menu.clock.16_4MHz_external.upload.speed=9600
328.menu.clock.16_4MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
328.menu.clock.16_4MHz_external.build.clkpr=-DOSC_PRESCALER=0x02
328.menu.clock.16_4MHz_external.build.clock_speed={build.f_cpu}
328.menu.clock.16_4MHz_external.build.f_cpu=4000000L
```

strange-v, 2020-11-08:
Thank you.

Maybe I'm wrong, but I don't see reasons why the bootloader may not work, with a clock prescaler MCU, will work at 4MHz from the beginning.

I'll test and let here know after new MCUs arrive, don't want to mess up the working node.

MCUdude, 2020-11-08:
The problem that the CLKPR register remains set to 0x02 even if you reset the microcontroller. This means that the register has to be cleared within the bootloader and set again in the sketch. The register is cleared in the bootloader here:

https://github.com/MCUdude/MiniCore/blob/96f4a0a843b1b66a9f167d4ba847d5e028c0db5b/avr/bootloaders/optiboot_flash/optiboot_flash.c#L517-L521

And the CLKPR register is set again here:

https://github.com/MCUdude/MiniCore/blob/96f4a0a843b1b66a9f167d4ba847d5e028c0db5b/avr/cores/MCUdude_corefiles/wiring.c#L615-L618

So when you try to upload a new sketch, the ATmega328P is actually running at 16 MHz again, and the baud rate defined in boards.txt is suddenly 4 times too slow. 

This will probably work: Select 16 MHz external oscillator, burn the bootloader, and select the custom 16/4 MHZ option. The code will run at 4 MHz, while the bootloader will run at 16 MHz.

strange-v, 2020-11-08:
Okay, now I see the problem.

The suggested approach will cause another issue. In the case of starting at 1.8V, MCU will try to run at 16MHz (when executing bootloader) but it'll fail due to the low voltage.

Thank you for pointing to exact places where prescaler is changing. Probably, I can fix the code a bit and rebuild the bootloader.

strange-v, 2020-11-17:
Decided, that according to the KISS principle, it'll be easier to buy required oscillators from Ali or other shops.

closed 2020-11-17 by strange-v

MCUdude, 2020-11-17:
> Decided, that according to the KISS principle, it'll be easier to buy required oscillators from Ali or other shop

Good idea. The bootloader is definitely the tricky part here, so a 4 MHz external crystal with no prescaler set is definitely the easy way out.