MCUdude/MiniCore #158: Not able to Export compiled binary file.

dharmik768, 2020-11-10:
Hello there,
I was trying to generate an .hex file for my recent project done on atmega328p but when I clicked export compiled binary file then then it shows error compilling and it shows the following output in verbos.
```
C:\software\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\software\Arduino\hardware -hardware C:\Users\tdhar\AppData\Local\Arduino15\packages -hardware C:\Users\tdhar\OneDrive\Documents\Arduino\hardware -tools C:\software\Arduino\tools-builder -tools C:\software\Arduino\hardware\tools\avr -tools C:\Users\tdhar\AppData\Local\Arduino15\packages -built-in-libraries C:\software\Arduino\libraries -libraries C:\Users\tdhar\OneDrive\Documents\Arduino\libraries -fqbn=MiniCore:avr:328:bootloader=no_bootloader,eeprom=keep,variant=modelP,BOD=disabled,LTO=Os_flto,clock=8MHz_external -ide-version=10808 -build-path C:\Users\tdhar\AppData\Local\Temp\arduino_build_139355 -warnings=none -build-cache C:\Users\tdhar\AppData\Local\Temp\arduino_cache_372941 -prefs=build.warn_data_percentage=75 -verbose E:\PCB\KRISENT\PROGRAMS\INdian eco 6\IND_eco6_V1.0.17\IND_eco6_V1.0.17.ino
C:\software\Arduino\arduino-builder -compile -logger=machine -hardware C:\software\Arduino\hardware -hardware C:\Users\tdhar\AppData\Local\Arduino15\packages -hardware C:\Users\tdhar\OneDrive\Documents\Arduino\hardware -tools C:\software\Arduino\tools-builder -tools C:\software\Arduino\hardware\tools\avr -tools C:\Users\tdhar\AppData\Local\Arduino15\packages -built-in-libraries C:\software\Arduino\libraries -libraries C:\Users\tdhar\OneDrive\Documents\Arduino\libraries -fqbn=MiniCore:avr:328:bootloader=no_bootloader,eeprom=keep,variant=modelP,BOD=disabled,LTO=Os_flto,clock=8MHz_external -ide-version=10808 -build-path C:\Users\tdhar\AppData\Local\Temp\arduino_build_139355 -warnings=none -build-cache C:\Users\tdhar\AppData\Local\Temp\arduino_cache_372941 -prefs=build.warn_data_percentage=75 -verbose E:\PCB\KRISENT\PROGRAMS\INdian eco 6\IND_eco6_V1.0.17\IND_eco6_V1.0.17.ino
Using board '328' from platform in folder: C:\Users\tdhar\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.6
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\tdhar\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.6
WARNING: Category '' in library UIPEthernet is not valid. Setting to 'Uncategorized'
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\cores\\MCUdude_corefiles" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\variants\\standard" "C:\\Users\\tdhar\\AppData\\Local\\Temp\\arduino_build_139355\\sketch\\IND_eco6_V1.0.17.ino.cpp" -o nul
"C:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\cores\\MCUdude_corefiles" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\variants\\standard" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\SendOnlySoftwareSerial-master" "C:\\Users\\tdhar\\AppData\\Local\\Temp\\arduino_build_139355\\sketch\\IND_eco6_V1.0.17.ino.cpp" -o nul
"C:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\cores\\MCUdude_corefiles" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\variants\\standard" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\SendOnlySoftwareSerial-master" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\rc-switch-master" "C:\\Users\\tdhar\\AppData\\Local\\Temp\\arduino_build_139355\\sketch\\IND_eco6_V1.0.17.ino.cpp" -o nul
"C:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\cores\\MCUdude_corefiles" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\variants\\standard" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\SendOnlySoftwareSerial-master" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\rc-switch-master" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\libraries\\EEPROM\\src" "C:\\Users\\tdhar\\AppData\\Local\\Temp\\arduino_build_139355\\sketch\\IND_eco6_V1.0.17.ino.cpp" -o nul
"C:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\cores\\MCUdude_corefiles" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\variants\\standard" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\SendOnlySoftwareSerial-master" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\rc-switch-master" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\libraries\\EEPROM\\src" "C:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\SendOnlySoftwareSerial-master\\SendOnlySoftwareSerial.cpp" -o nul
"C:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\cores\\MCUdude_corefiles" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\variants\\standard" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\SendOnlySoftwareSerial-master" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\rc-switch-master" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\libraries\\EEPROM\\src" "C:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\rc-switch-master\\RCSwitch.cpp" -o nul
Generating function prototypes...
"C:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\cores\\MCUdude_corefiles" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\variants\\standard" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\SendOnlySoftwareSerial-master" "-IC:\\Users\\tdhar\\OneDrive\\Documents\\Arduino\\libraries\\rc-switch-master" "-IC:\\Users\\tdhar\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.0.6\\libraries\\EE [... cut ...]
C:\Users\tdhar\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.6/scripts/delete_merged_output.bat false C:\Users\tdhar\AppData\Local\Temp\arduino_build_139355\IND_eco6_V1.0.17.ino.with_bootloader.hex 
cmd /C copy C:\Users\tdhar\AppData\Local\Temp\arduino_build_139355\IND_eco6_V1.0.17.ino_atmega328p_8000000L.lst E:\PCB\KRISENT\PROGRAMS\INdian eco 6\IND_eco6_V1.0.17 
The system cannot find the file specified.
Error compiling.
```
Following configerations are chosen 
[image]

and when I tried to export compiled binary file by selecting arduino uno as a board then it sucesfully generated .hex file in sketch location.

MCUdude, 2020-11-20:
Can you try to do a manual install to see if that version works? I've recently pushed a fix for this, but I've not done a new release yet.

MCUdude, 2020-12-02:
v2.0.8 is now available through boards manager, which has this bug fixed.

closed 2020-12-02 by MCUdude