MCUdude/MiniCore #159: Booting and flashing with custom clock prescaler

martincizek, 2020-11-25:
Follow up on #157 and #147:
* What's the reason for having the [prescaler hardcoded to `1` during boot](https://github.com/MCUdude/optiboot_flash/blob/master/optiboot_flash.c#L520)? 
* What's so special about 1MHz clock? I know it should be the ATMega328p factory default - 8 MHz internal oscillator with CKDIV8 fuse sets the *initial power-on prescaling* to 1/8, but this doesn't sound like a reasonable exception to the rule...

I want to use a MCU with an external 16MHz crystal (good reasons for it:)), when the voltage only allows for 8 MHz. I don't want to overclock the MCU even temporarily, so I need to program the CKDIV8 fuse, which ensures initial prescaling by 8. Then I need to change the prescaler to 2 ASAP, ideally before eventual flashing over UART. I was looking for a bootloader that supports this out of the box. And until examining the sources, I actually thought MiniCore supports this.

I guess the reason for hardcoding prescaler to 1 might be related to configuration straightforwardness (`OSC_PRESCALER` makes effect without burning the bootloader). But still, would it make sense to introduce initial prescaler value as a setting for advanced users?

MCUdude, 2020-11-25:
> What's the reason for having the prescaler hardcoded to 1 during boot?

The reason why is because the prescaler bits aren't cleared when a reset occurs. If the user application sets the prescaler to 2, the bootloader will only work once. You'll have to power cycle the chip to make the bootloader work again.

The bootloader has no idea what's going on in the user application.

* If you have a 16 MHz crystal and uses the "default", 16 MHz bootloader, the first upload will happen with a prescaler of 1, since the bootloader has no idea that you want to divide the clock down by a factor of two.
* Uploading goes well, and the main clock is divided down to 8 MHz and everything works fine.
* You want to re-flash the microcontroller. You're uploading using the bootloader again, but since the prescaler value is kept when a reset occurs, the bootloader is suddenly running at half speed. By setting the prescaler to 1 again makes sure the bootloader always runs at a known main clock speed, regardless of what's done in the user application.

I'm not using the prescaler to get a 1 MHz clock because this can be done in the fuse bits (CLKDIV8).

> But still, would it make sense to introduce initial prescaler value as a setting for advanced users?

Again, not possible, since the pre-compiled bootloader has no idea what the"user prescaler is".

In your special case, I suggest you compile your own bootloader where you hardcode the prescaler to be something known, in your case, 2.

MCUdude, 2020-11-25:
You can try these. These are bootloaders that expect the MCU to run at 8 MHz when the prescaler is set to 2. All I've done compared to the source code you looked at is that I've changed the hardcoded prescaler value from 1 to 2.
[ATmega328P_8000000L.zip](https://github.com/MCUdude/MiniCore/files/5599432/8000000L.zip)

martincizek, 2020-11-25:
Sorry, I edited the description to point out that I am not questioning the need for resetting CLKPR, just asking about the hardcoded value.

> I'm not using the prescaler to get a 1 MHz clock because this can be done in the fuse bits (CLKDIV8).

Are you sure about this? If I understand the docs correctly, CLKDIV8 just sets the initial value of prescaler during power-on.[^1] It has no further effect, you can change the prescaler whenever you want and it will last until power-cycling the chip. Which would mean that all the mentioned issues about not-resetting the prescaler apply when 1 MHz is used.
I guess the prescaler should be reset to 8 to match the non-1 MHz behavior. I can be wrong indeed. :)

> Again, not possible, since the pre-compiled bootloader has no idea what the"user prescaler is".

Actually this how my question should have been asked. I.e. how feasible is extending the number of precompiled variants? Or whether it makes sense to somehow support users in creating their own variants... Currently, there is sort-of support for configuring LED blinking, so that gave me confidence to ask about the prescaler. :)

I don't have any problem compiling my version, I'm not a fan of maintaining forks for changes that can be offered to the author. What about a preprocessor macro like this to at least externalize the constant and allow passing it from outside? The suggestion about 1 MHz already included, I can send a PR...
``` C
#ifndef OSC_START_CLKPR
#define OSC_START_CLKPR (F_CPU == 1000000L ?  0x3 : 0)
#endif

// OSC_START_CLKPR would then be used instead of hardcoded 0
```

[^1]: ATmega328p datasheet, chapter 8.12.2: _The CKDIV8 fuse determines the initial value of the CLKPS bits. If CKDIV8 is unprogrammed, the CLKPS bits will be reset to “0000”. If CKDIV8 is programmed, CLKPS bits are reset to “0011”, giving a division factor of 8 at start up._

martincizek, 2020-11-25:
> You can try these. These are bootloaders that expect the MCU to run at 8 MHz when the prescaler is set to 2. All I've done compared to the source code you looked at is that I've changed the [...]

Wow, appreciate your helpfulness! Will try it on weekend. I can also verify the suspected 1 MHz issue if it helps...

martincizek, 2020-11-30:
I didn't manage to play with the HW yet, but I've at least explored the sources more thoroughly.

Ref my suggedtion on
> `#define OSC_START_PRESCALER (F_CPU == 1000000L ?  0x3 : 0)`

... I've noticed that there are also definitions for `1MHz_external`, which would suffer the "unreliable CLKPR" issues above, but with this "fix", they will be reliably broken. :)

So it seems the only comprehensive solution is to include explicit initial prescaler to the precompiled bootloaders. I believe I can do it correctly, it is matter of:
1. Externalizing initial prescaler to a Macro and making the `Makefile` recognize it
2. Changing the filename pattern in Makefile for MCUs supporting prescaling
3. Make the `makeall` script generate the required combinations (this would be a longer description, but I've figured out a clean way how to do it without messing up the current definition of combinations and without generating unnecessary number of files at the same time).
4. Introduce a param in `boards.txt` and adjust the according bootloader filename patterns.
5. As a proud PlatformIO user, I don't know how the bootloader filename is generated (does it somehow use boards.txt too?), so this is something I would need help with.

So it seems to be quite a few changes given that it addresses relatively small issue. On the other hand, the behavior will then be consistent regardless of the chosen clock. 

So the question is: is it something you would consider if I made it and submitted a PR?

The low-hanging fruit would be just to externalize the value of initial prescaler in a macro...

martincizek, 2020-12-06:
Having a look at [PlatformIO's bootloader filename selection mechanism](https://github.com/platformio/platform-atmelavr/blob/v3.0.0/builder/bootloader.py#L34), it seems the file format has a generally accepted structure. The only property available for encoding the prescaler is probably the MCU name, which is ugly, although it makes some sense.

Will submit a PR for externalising the boot-time prescaler value in order to allow building custom bootloaders within this project. I guess it is always better to stick to a bootloader from the corresponding core and this should make it easier.

martincizek, 2020-12-08:
> You can try these. These are bootloaders that expect the MCU to run at 8 MHz when the prescaler is set to 2. All I've done compared to the source code you looked at is that I've changed the [...]

Tried it out, thanks again. I think these were compiled with CLKPR=2, which is prescaling factor 4.

For the sake of completeness (if anybody wants to try as well), these should be bootloaders with prescaler 2 (CLKPR=1):
[ATmega328P_8000000L.zip](https://github.com/MCUdude/MiniCore/files/5656386/8000000L.zip).

My current intention is to set up a build-only repo (i.e. not a fork, just CI+CD), which would build and offer PlatformIO package with custom bootloaders. The usage in PlatformIO would then be
1. use MiniCore as usual
2. include the "extra bootloaders" package using `platform_packages`
3. use overriden `board_bootloader.file`

This approach should allow to stick to MiniCore without the need to bloat it with plenty of precompiled bootloaders. Hope there won't be any blocker in Platformio package system. :)

MCUdude, 2022-08-19:
I'm closing this issue in favor of https://github.com/MCUdude/optiboot_flash/pull/7. And I'll look at this PR soon!

closed 2022-08-19 by MCUdude