MCUdude/MiniCore #160: Not able to upload code into atmega 328PB

bhaveshk023, 2020-11-27:
Was trying to upload code into Atmega 328PB using ISP method. Successfully burn the bootloader but was unable to burn the code. IDE was flashing below error.
avrdude: verification error, first mismatch at byte 0x0000
        0x00 != 0x0c
avrdude: verification error; content mismatch
Please kindly help me to solve this issue.

bhaveshk023, 2020-11-30:
Hey, Please help me out. I have done connections as per minimal setup and is continuously showing the above mentioned error. I am able to burn the bootloader but its flashing the error when uploading the code.

MCUdude, 2020-11-30:
If you search for that error you can see what others have done to solve it. There is nothing wrong with the bootloader itself. Also, you don't really have to use the bootloader either. You can use your programmer to upload code if that works out for you.

bhaveshk023, 2020-11-30:
Thank you for the reply. I have tried burning code directly without bootloader but was getting the same error. Now I tried using the UART method and the code was uploaded successfully. I don't get it why I was not able to burn the code using ISP method.

MCUdude, 2020-11-30:
> I have tried burning code directly without bootloader but was getting the same error.

This indicated that it's either something wrong with your programmer or more likely, an unreliable hardware setup. Have you wired it up on a breadboard, or do you have a dedicated PCB for the 328PB?

bhaveshk023, 2020-12-01:
I have an dedicated PCB for it. At first I was not able to burn the bootloader also then I changed the capacitor value from 100nf to 1000nf between VCC and GND then was able to put the bootloader successfully but had problem for uploading the code.

MCUdude, 2022-08-19:
Can you post the entire Avrdude output from when you're trying to upload code to the 328PB?

> At first I was not able to burn the bootloader also then I changed the capacitor value from 100nf to 1000nf between VCC and GND then was able to put the bootloader successfully but had problem for uploading the code.

This indicated that there are issues with your PCB or routing. I've designed 328PB based boards without any issues, so it's absolutely possible.

mcuee, 2022-12-23:
In any case, MiniCore optiboot_flash bootloader works fine with ATmega328PB. So this issue can be closed as the issue may be more HW related.

closed 2023-01-15 by MCUdude