MCUdude/MiniCore #161: Not able to burn Bootloader 328pb

ajjack50n, 2020-11-28:
Hi

I am pretty new to this but I am trying to burn a Bootloader to a 328PB

I have wired up the minimal circuit prescribed using a breadboard and a SA663 adapter and checked - it looks good.
Only diff is that I  am not using an external clock and not wired UART yet.

Board: ARmega328
Clock: Intenral 8mhz - I have tried others.
BOD: disabled
EEPROM: retained
Variant: 328PB
Bootloader Yes(UART0)
Programmer USBasp (MiniCore)

When I try to burn the Bootloader from within Arduino IDE, I get initialisation failed.. double check connections.
It also suggest that I should try -F but I am not certain how to do that? Presumably I can run this same command from terminal?

Any help really appreciated - as I say, I suspect I am doing something wrong but can’t figure it out.

Thanks

Andy

MCUdude, 2020-11-28:
Turn on verbose upload output in the Arduino IDE settings and post the actual output here.

Is the ATmega328PB a factory blank unit? If so, it's running off its internal oscillator, which is too slow for the USBasp. You can try the `USBasp slow` option and see if it makes a difference.

ajjack50n, 2020-11-28:
hey.. thanks for responding

It is a blank unit. 

I have tried both the USB options, below is the output from the slow option. I am not certain turning on verbose mode revealed any more. I presume you mean the options for verbose mode under preferences in the Arduino IDE? Otherwise I ran with same setting in the above post.  Output pasted below. 

Further below, I realised I could execute in the terminal with -F switch, and I have pasted the result.
Looking at these logs, to me, it looks like there is something fundamental wrong - like my wiring but I have double checked the same.

> Arduino: 1.8.13 (Mac OS X), Board: "ATmega328, Yes (UART0), EEPROM retained, 328PB, BOD  disabled, LTO disabled, Internal 8 MHz"
> 
> /Users/andy/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/andy/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.7/avrdude.conf -v -patmega328pb -cusbasp -Pusb -B32 -e -Ulock:w:0x3f:m -Uefuse:w:0b11110111:m -Uhfuse:w:0b11010110:m -Ulfuse:w:0xe2:m 
> 
> avrdude: Version 6.3-20190619
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "/Users/andy/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.7/avrdude.conf"
>          User configuration file is "/Users/andy/.avrduderc"
>          User configuration file does not exist or is not a regular file, skipping
> 
>          Using Port                    : usb
>          Using Programmer              : usbasp
>          Setting bit clk period        : 32.0
>          AVR Part                      : ATmega328PB
>          Chip Erase delay              : 9000 us
>          PAGEL                         : PD7
>          BS2                           : PC2
>          RESET disposition             : dedicated
>          RETRY pulse                   : SCK
>          serial program mode           : yes
>          parallel program mode         : yes
>          Timeout                       : 200
>          StabDelay                     : 100
>          CmdexeDelay                   : 25
>          SyncLoops                     : 32
>          ByteDelay                     : 0
>          PollIndex                     : 3
>          PollValue                     : 0x53
>          Memory Detail                 :
> 
>                                   Block Poll               Page                       Polled
>            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>            eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
>            flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
>            lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
>            hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
>            lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
>            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
>            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
>            efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
> 
>          Programmer Type : usbasp
>          Description     : USBasp, http://www.fischl.de/usbasp/
> 
> avrdude: set SCK frequency to 16000 Hz
> avrdude: error: program enable: target doesn't answer. 1 
> avrdude: initialization failed, rc=-1
>          Double check connections and try again, or use -F to override
>          this check.
> 
> 
> avrdude done.  Thank you.
> 
> Error while burning bootloader.

And when ran in terminal with switch...

> avrdude version 6.3-20190619, URL: <http://savannah.nongnu.org/projects/avrdude/>
[... 31 lines cut ...]
>            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>            eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
>            flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
>            lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
>            hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
>            lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
>            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
>            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
>            efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
> 
>          Programmer Type : usbasp
>          Description     : USBasp, http://www.fischl.de/usbasp/
> 
> avrdude: set SCK frequency to 16000 Hz
> avrdude: error: program enable: target doesn't answer. 1 
> avrdude: initialization failed, rc=-1
> avrdude: AVR device initialized and ready to accept instructions
> avrdude: Device signature = 0x000000 (retrying)
> avrdude: Device signature = 0x000000 (retrying)
> avrdude: Device signature = 0x000000
> avrdude: Yikes!  Invalid device signature.
> avrdude: Expected signature for ATmega328PB is 1E 95 16
> 
> avrdude done.  Thank you.

ajjack50n, 2020-11-29:
Hey.. stand down... I took apart and rewired from scratch. Not certain what was different this time but it works now!:)

Sorry to bother but thanks for help / response. Will close this non issue!

closed 2020-11-29 by ajjack50n

MCUdude, 2020-11-29:
Great to hear you figured it out! A communication error like this is very rarely MiniCore's fault. It's almost always a hardware related issue.