MCUdude/MiniCore #162: Cannot upload sketch via UART after power off Atmega88

raymondle, 2020-12-05:
Hi everyone,
I had an issue when trying upload a sketch in Arduino to Atmega88PA-AU and here below is reproduce steps:
1. Setup connection between usb ttl cp2102 -> Arduino mini -> Atmega88
2. Burn bootloader via Arduino as ISP success.
3. Remove Arduino.
4. Setup USB TTL connection with Atmega88 via UART and follow minimal setup from https://github.com/MCUdude/MiniCore#minimal-setup

[image]

5. Do simple blink code and upload sketch with successful.
6. Edit code and upload again but have an error
```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xxx\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8/avrdude.conf"

         Using Port                    : COM9
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

An error occurred while uploading the sketch
```
7. Wait a few minutes and click upload again with success or simply do power off Atmega and power up again after a few seconds.

And here are some addition details:

- Board: Atmega88
- Bootloader: Yes
- Clock: External 16Mhz
- Compile LTO: disable
- Variant: 88 / 88PA
- BOD: BOD 1.8V
- Programmer: AVR ISP (MiniCore)
- Arduino 1.8.5 with Windows 10 64Bit Latest Update

So do I missing something or do anything wrong? Please help me figure out and thank you so much!

MCUdude, 2020-12-05:
Issues like this are almost always caused by some hardware issue. One issue I can see in your Fritzing drawing is the pullup resistor on the reset line. It is supposed to be connected to the microcontroller reset pin, not to the DTR line. Just wire it to the other end of the capacitor.

raymondle, 2020-12-05:
> Issues like this are almost always caused by some hardware issue. One issue I can see in your Fritzing drawing is the pullup resistor on the reset line. It is supposed to be connected to the microcontroller reset pin, not to the DTR line. Just wire it to the other end of the capacitor.

Firstly, thanks for you support. But in fact, my USB TTL have DTS and RST in the same port like DTS/RST, so in my Fritzing, DTR and RST is the same.
!(image)[https://bizweb.dktcdn.net/thumb/large/100/190/540/products/usb-to-ttl-cp2102-7f3402f8-0fc1-4f9a-a35e-bccb0c17816f.jpg?v=1547177131307]

MCUdude, 2020-12-05:
> DTR and RST is the same and i'm sorry about this missing.

No, that's not what I meant. What I meant was that the pullup resistor isn't connected to the reset pin on the microcontroller

raymondle, 2020-12-06:
> > DTR and RST is the same and i'm sorry about this missing.
> 
> No, that's not what I meant. What I meant was that the pullup resistor isn't connected to the reset pin on the microcontroller

Hi @MCUdude, I'm following your instruction but no luck 😢 . I found a problem at capacitor 100nF at Reset Line, If I remove it and put down again in a few second, I could upload success.

[image]

raymondle, 2020-12-09:
Hi @MCUdude, thank you sooo much. I have resolved this issue because my breadboard has broken, the +VIN and GND on two sides not connect together. So that with my pull up resistor not working as well!

closed 2020-12-09 by raymondle

MCUdude, 2020-12-09:
Great to hear you figured it out!