MCUdude/MiniCore #164: Using Minicore version 2.0.8 breaks the use of the library Arduino_Vcc

ericvb, 2020-12-09:
Using Minicore version 2.0.7 with this library https://github.com/Yveaux/Arduino_Vcc, in an arduino sketch works without any problem.
When using Minicore version 2.0.8 with this same lib in the same arduino sketch produces these compiler errors:

```
"C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_851041\\libraries\\VccBatteryPercent\\Vcc.cpp.o"
In file included from C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.h:27:0,
                 from C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:24:
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp: In member function 'float Vcc::Read_Perc(float, float, boolean)':
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:133:22: error: 'l' was not declared in this scope
   typeof (low) _l = (l);           \
                      ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:133:22: note: suggested alternative: '_l'
   typeof (low) _l = (l);           \
                      ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:134:23: error: 'h' was not declared in this scope
   typeof (high) _h = (h);          \
                       ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:134:23: note: suggested alternative: '_h'
   typeof (high) _h = (h);          \
                       ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:135:35: error: expected ';' before '}' token
   _x < _l ? _l : _x > _h ? _h :_x })
                                   ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~

```

I opened a issue in his github project : https://github.com/Yveaux/Arduino_Vcc/issues/10, but the author closed it and redirected the problem to Minicore
_This library works with regular arduino IDE versions. Furthermore a change of the minicore library apparently broke the constrain function._
_I would suggest to contact the author of minicore._

An idea of the problem?
Probably related to the latest changes in these 'unsafe' macros...

MCUdude, 2020-12-09:
> Probably related to the latest changes in these 'unsafe' macros...

That's true. I accidentally introduced a typo in the `constrain()` macro. A new release will probably be available tomorrow. I just have to figure out another unrelated, potential bug.

MCUdude, 2020-12-10:
MiniCore v2.0.9 is now available through boards manager. Please retry with this version and report back if it works or not 🙂

closed 2020-12-10 by MCUdude

ericvb, 2020-12-14:
@MCUdude it works! :-)
Thanks

commit 7ecaecaa (2020-12-10): Add 2.0.9 to boards manager
Closes #164