MCUdude/MiniCore #166: How to insert a custom fuse (Clock output) on 328

Betacarotin, 2020-12-13:
I would like to set my 328 into these fuses: low 0x82 high: 0xD9 extended 0xff

This gives mne an internal 8 Mhz clock, that is remapped to to also clock output! With that I can use a lot of other hardware also without external crystals. It also gives me superfast booting. 

I dont find these options in the menu, and I have no idea how to add features... 
is there a easy way to get these fuses with minicore?

MCUdude, 2020-12-13:
You can use PlatformIO to do this. Or modify the boards.txt file.

But PlatformIO is definititly the most elegant solution.
https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md

closed 2020-12-13 by MCUdude

Betacarotin, 2020-12-13:
can you give me a hint how to modify the borads.txt?
Whenever i try to get my 
`328.bootloader.low_fuses=0x82
328.bootloader.high_fuses=0xD9
328.bootloader.extended_fuses=0xff`

in there, I am breaking something and the outcome is ..at least a error :-/ 

I also tried PlatformIO on Atom...I dont understand a single bit of it.. what is that? Another text editor?

Betacarotin, 2020-12-13:
`# Menu options
menu.clock=Clock
menu.BOD=BOD
menu.eeprom=EEPROM
menu.LTO=Compiler LTO 
menu.variant=Variant
menu.bootloader=Bootloader

#############################
#### ATmega328/A/P/PA/PB ####
#############################

# General
328.name=ATmega328-Micha special
328.upload.tool=avrdude
328.upload.maximum_data_size=2048
328.bootloader.tool=avrdude
328.bootloader.unlock_bits=0x3f
328.bootloader.lock_bits=0x0f
328.build.core=MCUdude_corefiles
328.build.board=AVR_ATmega328
328.build.bootloader_led=B5
328.build.extra_flags={build.clkpr}

# Upload port select
328.menu.bootloader.uart0=Yes (UART0)
328.menu.bootloader.uart0.upload.maximum_size=32256
328.menu.bootloader.uart0.upload.protocol=arduino
328.menu.bootloader.uart0.upload.port=UART0
328.menu.bootloader.uart0.build.export_merged_output=true
328.menu.bootloader.uart0.bootloader.high_fuses=0xd9
328.menu.bootloader.uart0.bootloader.file=optiboot_flash/bootloaders/{build.mcu}/{build.clock_speed}/optiboot_flash_{build.mcu}_{upload.port}_{upload.speed}_{build.clock_speed}_{build.bootloader_led}.hex

328.menu.bootloader.no_bootloader=No bootloader
328.menu.bootloader.no_bootloader.upload.maximum_size=32768
328.menu.bootloader.no_bootloader.build.export_merged_output=false
328.menu.bootloader.no_bootloader.bootloader.high_fuses=0b1101{bootloader.eesave_bit}111
328.menu.bootloader.no_bootloader.bootloader.file=empty/empty.hex

# EEPROM

# Variants
328.menu.variant.modelP=328P / 328PA
328.menu.variant.modelP.bootloader.cksel_bits=0111
328.menu.variant.modelP.bootloader.cfd_bit=1
328.menu.variant.modelP.build.variant=standard
328.menu.variant.modelP.build.mcu=atmega328p

328.menu.variant.modelNonP=328 / 328A
328.menu.variant.modelNonP.bootloader.cksel_bits=0111
328.menu.variant.modelNonP.bootloader.cfd_bit=1
328.menu.variant.modelNonP.build.variant=standard
328.menu.variant.modelNonP.build.mcu=atmega328

328.menu.variant.modelPB=328PB
328.menu.variant.modelPB.bootloader.cksel_bits=1111
328.menu.variant.modelPB.bootloader.cfd_bit=0
328.menu.variant.modelPB.build.variant=pb-variant
328.menu.variant.modelPB.build.mcu=atmega328pb

# Brown out detection
328.menu.BOD.2v7.bootloader.extended_fuses=0xff

# Compiler link time optimization
328.menu.LTO.Os=LTO disabled
328.menu.LTO.Os.compiler.c.extra_flags=
328.menu.LTO.Os.compiler.c.elf.extra_flags=
328.menu.LTO.Os.compiler.cpp.extra_flags=
328.menu.LTO.Os.ltoarcmd=avr-ar

328.menu.LTO.Os_flto=LTO enabled
328.menu.LTO.Os_flto.compiler.c.extra_flags=-Wextra -flto -g
328.menu.LTO.Os_flto.compiler.c.elf.extra_flags=-w -flto -g 
328.menu.LTO.Os_flto.compiler.cpp.extra_flags=-Wextra -flto -g
328.menu.LTO.Os_flto.ltoarcmd=avr-gcc-ar

# Clock frequencies

328.menu.clock.8MHz_internal=Internal 8 MHz
328.menu.clock.8MHz_internal.upload.speed=38400
328.menu.clock.8MHz_internal.bootloader.low_fuses=0x82
328.menu.clock.8MHz_internal.build.clkpr=
328.menu.clock.8MHz_internal.build.clock_speed={build.f_cpu}
328.menu.clock.8MHz_internal.build.f_cpu=8000000L`

Betacarotin, 2020-12-13:
This lets me burn the bootloader with my fuses, but i cant upload any sketch with this over the arduino as ISP function afterwards

MCUdude, 2020-12-13:
`328.menu.clock.8MHz_internal.bootloader.low_fuses=0x82` should enable CKOUT when the internal 8 MHz oscillator is in use. If you're having upload problems it's probably something else. But you haven't posted any errors, so it's impossible for me to say.

Betacarotin, 2020-12-13:
yes it enables it. 
What I posted above is my complete boards.txt. 
The bootloader burns in, but any upload of sketches throws an error ( i did not record) 
Now I messed up all mt avrdude.config files too much (they all throw now the: avrdude: Expected signature for ATMEGA328P is 1E 95 0F Double check chip, or use -F to override this check.
error. 

let me get rid of my installation and begin new from scratch, I guess it will work than :-) 
Thanks!