MCUdude/MiniCore #170: Can not program the micro since V2.0.6 in Sloeber

ospilos, 2021-02-04:
Since 2.0.6. the .hex can not be programmed into the micro, at least in Sloeber. 
The cause is that the only {build.project_name}.hex is created, but AVRdude is always invoked with {build.project_name}.with_bootloader.hex, that does not exist.
Changing the "{build.project_name}.with_bootloader.hex" to "{build.project_name}.hex" in file platform.txt solves this problem.
The problem did not exist in MiniCore V2.0.5., but aparently is present since V2.0.6 up to V2.0.9.

Best regards, Tom

MCUdude, 2021-02-05:
This is something you'll have to report to the Sloeber author. From v2.0.6, I replaced "{build.project_name}.hex" with "{build.project_name}.with_bootloader.hex" so that when you have selected a bootloader in the Tools menu and upload using a programmer, a hex file with the bootloader file is included. I have no plans to change this.

MiniCore is not designed to work with Sloeber. I'll recommend PlatformIO if you want something else than Arduino IDE

closed 2021-02-05 by MCUdude

ospilos, 2021-02-05:
Thanks for letting me know.
Cheers! Tom