MCUdude/MiniCore #171: The problem of using analog inputs A6, A7

naro91, 2021-02-15:
When you try to use the digital outputs 6, 7 (number of arduino pins) and analog inputs, A6 and A7, at the same time, the pin assignment is redefined, since the numbers 6, 7 are defined in the library under the constants A6 and A7.

Example:
pinMode(7, OUTPUT);
pinMode(A7, INPUT); overrides the assignment of pin 7, since under the constant A7 = 7

digitalWrite(7, HIGH); does not work already, since pin 7 is in ADC7 analog input mode

is it a library or usage problem?

bigjosh, 2021-02-15:
On this chip, analog inputs 6 and 7 are analog only - they can not be digital. 

This means that they are always in "INPUT" mode. You do not need to set them to input mode. 

To read one of these input, you should be able to just do...

`int x = analogRead( A7);`

This should work even if you have previously done a `pinMode(7,OUTPUT);` since they are different pins.

Can  you check and see if this works and report back?

bigjosh, 2021-02-15:
...and to be clear, if you do a `digitalRead(7)` you will get the digital input on digital pin 7 not the analog pin. The only thing you can do on pins `A6` and `A7` is `analogRead()`.

(It would be nice if the the `pinmode` and `digital` functions made a compiler error if you tried to call them with `A6` or `A7`, but they don't.)

naro91, 2021-02-24:
Thank you very much for the answer!
Yes, I tried the option you suggested and it worked.

bigjosh, 2021-02-24:
Closes #171

closed 2021-02-24 by MCUdude