MCUdude/MiniCore #172: round() macro produces wrong results

ThFischer, 2021-02-18:
The `round()` macro
```
#define round(x)     ({ typeof (x) _x = (x); _x >= 0 ? (long)_x + 0.5 : (long)_x - 0.5; })
```
produces wrong results:

```
Serial.println(round(1.0f)); // ==> "1.50"
```

Proposal for a fixed `round()`:
```
#define round(x) ({ (typeof (x)) (long) (x + (x >= 0 ? 0.5 : -0.5)); })
```

MCUdude, 2021-02-18:
Hi!

This was fixed here a while ago:

https://github.com/MCUdude/MCUdude_corefiles/commit/8913fb360e3911f5d142c15dec921793d32a5e27#diff-b6ae690c4cd20807af23c0682cb5a4edbcf0a67e814501d83ba154889ecdc1bc

but I haven't done a new release of MiniCore yet. I should probably do this soon.

ThFischer, 2021-02-18:
Hi Hans,
Thanks! That would be very nice, since quite a lot of developers may affected since MiniCore is used by PlatformIO.

MCUdude, 2021-02-20:
Fixed in v2.1.0

closed 2021-02-20 by MCUdude