MCUdude/MiniCore #173: Customize bootloader for 328PB

mahtoyy, 2021-02-18:
Hey,
I am writing about an issue I encountered recently. I am a hobbyist, who recently started dealing with AVR in depth.
I have a project where I want to flash code through UART1 on a 328PB board. The issue is that I am using rs485 which is half-duplex and needs to pull a pin to HIGH state in order to enable transmit. I tried just to pull the pin high in the putch() function and then set it low again but unfortunatelly it did not work. When I try to flash code the device cannot respond.
Do you have any suggestions, maybe changing some registers or something like that?
I would be very glad if you can help me through!
Thanks in advance!

MCUdude, 2021-02-18:
I'm not the author of Optiboot, I've just modified the source code and created a build script that throws out hex files for all targets for most clock and baud rate combinations. I'll recommend you open an issue over at [the official Optiboot repo](https://github.com/Optiboot/optiboot).

mahtoyy, 2021-02-18:
Thank you for your answer! I will!

closed 2021-02-18 by mahtoyy