MCUdude/MiniCore #174: Problem with rc-switch with the release 2.1.0

rolie31, 2021-02-21:
Hi,

I use mini core with my atmega328p, 1mhz internal clock. Thx :)

With all the version I test since about 1 year, all works fine with the rc-switch library (https://github.com/sui77/rc-switch)

With the last release, 2.1.0, the event are not raised. If i downgrade to 2.0.9, I have event that are raised. 

I can make some test if it helps to resolve this issue or give more information if needed

MCUdude, 2021-02-21:
Does it work if you use a higher clock frequency than 1 MHz?

I don't really know anything about the rc-switch library. Is it a sender or receiver that doesn't work? what rc-switch function is it that doesn't "raise an event"?

rolie31, 2021-02-21:
I will test with higher clock frequency in the next days (I use 1mhz because it consummes less current).

Here is the function I use

RCSwitch mySwitch = RCSwitch();
...
  if (mySwitch.available()) { ....}

With all the previous version, when a rf 433 message is received (motion detected by motion detector), this available() function return true.
With the 2.1.0, it never return true

MCUdude, 2021-02-21:
I'll assume you've installed v2.1.0 using boards manager? If so, I have just updated the compiler version fro avr-gcc 5.4.0 to 7.3.0, which is the latest AVR based Arduinos have available.

If you look in the library source code, and you try replacing all

```
#if not defined( RCSwitchDisableReceiving )
```

with 

```
#ifndef RCSwitchDisableReceiving

And see of it makes a difference?

rolie31, 2021-02-22:
Yes I have used the board manager to update the 2.1.0.

I have replaced all with #ifndef RCSwitchDisableReceiving in RCSwitch library as you suggest.
It's not working with 2.1.0
With or without this replacement, it's working with 2.0.9

I try to burn bootloader 8 mhz internal but I have the same issue with 2.1.0. 
I also try to burn booloader with 16hz external clock and I also have the issue.

With or without the replace, it's not working.

MCUdude, 2021-02-22:
Thanks for testing. Can you try to downgrade to v2.0.9 again, make sure it works again and then replace the contents of wiring.c with this?https://github.com/MCUdude/MiniCore/blob/a737255ba484b8cb35a2186e5f9391de31b76c5e/avr/cores/MCUdude_corefiles/wiring.c

If you turn on verbose compilation in the IDE settings you can easily figure out the path where wiring.c is located.

The only major difference from v2.0.9 to v2.1.0 is timing-related improvements, increased accuracy for millis(), micros() and delay. It would be interesting to see if this was to root cause or not.

Thanks!

rolie31, 2021-02-22:
I will try this tomorrow

rolie31, 2021-02-23:
Hi,
I try it but with this file I do not have the issue.

MCUdude, 2021-02-23:
> I try it but with this file I do not have the issue.

OK. So this proves that it has something to do with the latest wiring.c file. @cburstedde do you have any idea why this library doesn't work with the timing improvements you've implemented?

rolie31, 2021-02-23:
with this file or the other in 2.0.9, it works

MCUdude, 2021-02-23:
> I don't see how this prove that is something to do with wiring.c
with this file or the other in 2.0.9, it works

Sorry? I got the impression that the rc-switch library 
* Works with 2.0.9
* Does not work with 2.0.9 if you replace the contents of Wiring.txt with [this](https://github.com/MCUdude/MiniCore/blob/a737255ba484b8cb35a2186e5f9391de31b76c5e/avr/cores/MCUdude_corefiles/wiring.c).

Am I right?

rolie31, 2021-02-23:
It's working without and with the wiring.c file replaced in 2.0.9

rolie31, 2021-02-23:
Work with both

MCUdude, 2021-02-23:
OK, that clarify things a little. Can you try this too?

* Make sure you have 2.0.9 installed, and it is working
* Replace the contents of Arduino .h with the contents from this file: https://github.com/MCUdude/MiniCore/blob/a737255ba484b8cb35a2186e5f9391de31b76c5e/avr/cores/MCUdude_corefiles/Arduino.h
* Does it still works now?

rolie31, 2021-02-23:
- Make sure you have 2.0.9 installed, and it is working
**Working**
- Replace the contents of Arduino .h with the contents from this file: https://github.com/MCUdude/MiniCore/blob/a737255ba484b8cb35a2186e5f9391de31b76c5e/avr/cores/MCUdude_corefiles/Arduino.h
**OK**
- Does it still works now?
**NO**

MCUdude, 2021-02-23:
Good! This means that there's something with the Arduino.h file.

can you
* Keep the changes so it doesn't work with 2.0.9
* Add this code before you include rc-switch in your sketch:

```c++
#undef abs
#define abs(x)    __builtin_abs(x)
#include <RCSwitch.h> // Include library
```
* Does it work now?

rolie31, 2021-02-23:
It's not working. I also try this to be sure

```
#include "Arduino.h"
#include <avr/wdt.h>

#undef abs
#define abs(x)       __builtin_abs(x)

#undef min
#define min(a,b)     ({ typeof (a) _a = (a); typeof (b) _b = (b); _a < _b ? _a : _b;    })

#undef max
#define max(a,b)     ({ typeof (a) _a = (a); typeof (b) _b = (b); _a > _b ? _a : _b;    })

#undef round
#define round(x)     ({ typeof (x) _x = (x); _x >= 0 ? (long)_x + 0.5 : (long)_x - 0.5; })

#include <EEPROM.h>
#include <RCSwitch.h>

```

And, it's also not working.  Weird ...

rolie31, 2021-02-23:
I went back to 2.0.9 and delete code change in my sketch;

I edit directly arduino.h and make those 3 lines changes

```
#define min(a,b)     ({ typeof (a) _a = (a); typeof (b) _b = (b); _a < _b ? _a : _b; })
#define max(a,b)     ({ typeof (a) _a = (a); typeof (b) _b = (b); _a > _b ? _a : _b; })
#define round(x)     ({ typeof (x) _x = (x); _x >= 0 ? (long)(_x + 0.5) : (long)(_x - 0.5); })

```
and let this line `
#define abs(x)       __builtin_abs(x)`

Test it. 
**OK**

If I use this one
`#define abs(x)       ({ typeof (x) _x = (x); _x > 0 ? _x : -x; })`
**KO**

If I test this one 
`#define abs(x)       ({ typeof (x) _x = (x); _x > 0 ? _x : -_x; })
**OK**

I dont know why undef not working but your modification seems to work :)

MCUdude, 2021-02-23:
Thank you for testing. While I wrote the previous post I discovered a bug in the abs() macro:

``` c
// Won't work
#define abs(x) ({ typeof (x) _x = (x); _x > 0 ? _x : -x; }) // It's supposed to be -_x, not -x

// Will work
#define abs(x) ({ typeof (x) _x = (x); _x > 0 ? _x : -_x; }) // Typo corrected
```

I'll fix this very soon and release 2.1.1. This bug is very sneaky and difficult to trace down. Thank you for reporting, it helps alot!

rolie31, 2021-02-23:
You're welcome.
Thanks for all your work

MCUdude, 2021-03-01:
A new release, MiniCore v2.1.1 is now available through boards manager. The new version fixes the typo in the abs() macro that caused the issue you were seeing.

closed 2021-03-01 by MCUdude