MCUdude/MiniCore #177: compatibility with official Arduino AVR core?

greyltc, 2021-02-22:
If I write code works with the latest stable release of https://github.com/arduino/ArduinoCore-avr should I expect it to work here too?

MCUdude, 2021-02-22:
Hi!

> If I write code works with the latest stable release of https://github.com/arduino/ArduinoCore-avr should I expect it to work here too?

That is the goal, yes. This core has some optimizations that the official core doesn't have, and it's been designed to work with a wide variety of chips and clocks, not just the ATmega328P @ 16 MHz. But I'm always trying to keep this (and my other AVR Arduino cores) compatible with the official Arduino AVR core.

If you for some reason have compatibility issues, please submit an issue, and I'll look into it.

greyltc, 2021-02-22:
Good to know! Issue incoming ;-)

closed 2021-02-22 by greyltc