MCUdude/MiniCore #181: SERNUM register not defined

jashsheth, 2021-03-20:
Hi community,
I am trying to access the 10byte unique serial number in Atmega328pb, which resides in SERNUMn registers.
When simply trying to do a register read, I am not able to do it as it says that this register is not defined in the scope.
Also tried SIGROW.SERNUM0 but that to does not help.
Also how can I get the serial number completely in a 16bit int to be used in the code
Please help

Thanks

MCUdude, 2021-03-20:
The 328PB serial number is not defined as a register you can easily use line `SIGROW.SERNUM0`. You'll have to do sonething else. Read this post (end the entire thread if you're interested in the topic): https://www.avrfreaks.net/comment/1743391#comment-1743391

MCUdude, 2021-03-20:
I found this in the 328PB datasheet too:

> Reading the Signature Row from Software
To read the Signature Row from software, load the Z-pointer with the signature byte address given in the
following table and set the SIGRD and SPMEN bits in SPMCSR (SPMCSR.SIGRD and
SPMCSR.SPMEN). When an LPM instruction is executed within three CPU cycles after the
SPMCSR.SIGRD and SPMCSR.SPMEN are set, the signature byte value will be loaded in the
destination register. The SPMCSR.SIGRD and SPMCSR.SPMEN will auto-clear upon completion of
reading the Signature Row Lock bits or if no LPM instruction is executed within three CPU cycles. When
SPMCSR.SIGRD and SPMCSR.SPMEN are cleared, LPM will work as described in the Instruction set
Manual.

Note that you can't use `SPMCSR.SIGRD` notation like you can on the newer megaAVR-0 series. Here's the SPMCSR register definition:
``` c
#define SPMCSR  _SFR_IO8(0x37)
#define SPMEN   0
#define PGERS   1
#define PGWRT   2
#define BLBSET  3
#define RWWSRE  4
#define SIGRD   5
#define RWWSB   6
#define SPMIE   7
```

jashsheth, 2021-03-21:
Hi Hans,
Thank you for your quick response!!
I will go through the thread you mentioned, !
Also about the SPMCSR register, if I cant access with the ".",  do I have to access it like SPMCSR=00000100 to set the SIGRD bit??

Thanks a ton!!

MCUdude, 2021-06-06:
> Also about the SPMCSR register, if I cant access with the ".", do I have to access it like SPMCSR=00000100 to set the SIGRD bit??

You access the SPMCSR register as you'll access any other register on the 328PB.

closed 2021-06-06 by MCUdude