MCUdude/MiniCore #183: Modbus problem

kamil-bit, 2021-03-31:
When i want to use <ArduinoRS485.h> and <ArduinoModbus.h>
in example ModbusRTUClientToggle
https://www.arduino.cc/en/ArduinoModbus/ArduinoModbus
i get this 
\libraries\ArduinoRS485-master\src\RS485.cpp:181:18: error: 'SERIAL_PORT_HARDWARE' was not declared in this scope
 RS485Class RS485(SERIAL_PORT_HARDWARE, RS485_DEFAULT_TX_PIN, RS485_DEFAULT_DE_PIN, RS485_DEFAULT_RE_PIN);
                               
i set
atmega328
eeprom not retained
lto enabled
variant 328p 328pa
no bootloader

when i compile the same program on arduino uno or leonardo is well

MCUdude, 2021-03-31:
What happens if you add `#define SERIAL_PORT_HARDWARE Serial` before including the library?

kamil-bit, 2021-04-02:
Nothing again the same. I use 2.1.1 version of minicore

kamil-bit, 2021-08-05:
Naprawiono

kamil-bit, 2021-08-05:
Naprawiono

closed 2021-08-05 by kamil-bit