MCUdude/MiniCore #184: Bootloader on ATMega328PB randomly stops working

Tsamsiy, 2021-03-31:
Hi

Thanks for the amazing work! 
All my new projects run with your Core on my new favourite chip: the ATMega328PB.

Usually, the hardware setup consits of a dedicated PCB that directly contains the MCU or a selfmade Arduino Pro-Mini compatible module with this chipset.
The serial bootloader for UART0 is burned before assembly using a "Arduino as ISP" setup with QFN24 socket and matching external crystal.
Other setting are usually:
16MHz external [compact resonator (the ones that combine crystal and caps)]
BOD disabled
EEPROM retained
LTO enabled
PB variant

Everything works fine until i get this weird issue where the IDE would tell me that the upload was successful but the program on the chip would still be the old one.
Detaild outputs on the IDE's console revealed that a content verification mismatch error occured (sorry i don't have the log messages right now).
While serching for this error (can't remember souce), i read that re-flashing the bootloader would fixt it. 
Even though this occured on a different chip than i used, i tried it and it worked.... temorarily. After a random number of uploads while developing, it would lock out again.
The weird thing about this, is that this would occur again after a seemingly random number of uploads and would only prevent a new sketch from uploading.
The code that is already on the chip remains intact. I was not able to trace the issue to any specific reason or circumstance that would be able to cause something like that.
Also i've had the issue occur on a few entirely different PCB Layouts by now. I also tried different crystal frequencies like 8MHz and 20MHh and different supply voltages like 3,3V and 5V.
None of these options including heavy shielding with copper tape seem to prevent this from happening again.

Regarding the usage of the 20MHz resonator, i've had some trouble burning the bootloader to these. Uploading sketches is even more difficult.
The upload stops after a bit and i get "Programmer out of sync" errors.

I hope you have a clue to what is happening here..
If it helps you fix bugs or find bugs in my hardware that i didn't find, i can send you a sample of the used modules (including schematics/gerbers) for testing.

MCUdude, 2021-03-31:
Do you have a datasheet for the resonator you're using? The ATmega328PB (and ATmega324PB, its bigger brother) doesn't have a full swing oscillator to drive the resonator, so they are very picky on what they can drive.

I've screwed up on this myself. I designed a board for commercial use that used an ATmega324PB with an external, 8 MHz Murata regulator with 27pF internal capacitors. These boards would occasionally freeze, but it could take weeks or even months. Took me a while to figure out, and there were already hundreds of units sold before we figured out what it was and what caused it. I swapped the resonator out with one with 10pF capacitors and have been happy ever after.

MCUdude, 2021-03-31:
According to the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/40001906A.pdf) (chapter 11.3) The ATmega328PB is only rated to drive a maximum of 16 MHz crystal or resonator with a load between 12 and 22pF. If you want/need a higher clock frequency you should go with an external clock pulse fed into the XTAL1 pin instead (chapter 11.7)

Tsamsiy, 2021-03-31:
Thank you very much for the fast response!

I have been using the Murata CSTCE16M0V53-R0 , which has 15pF integrated caps, so i guess these are within range... but i guess not?
Here's the link to where i got them from: https://lcsc.com/product-detail/Ceramic-Resonators_Murata-Electronics-CSTCE16M0V53-R0_C32180.html

What do you mean by "freeze"?
Mine didn't exactly just "freeze". The installed program did continue to work as its supposed to. Just the upload of a new sketch was bugged.
But i don't understand why reflashing the bootloader would, at least temporarily, fix it.

MCUdude, 2021-03-31:
> What do you mean by "freeze"?

It seemed that the application stopped running. It was not possible to interface with the device. Since the device didn't have a reset button, I had to power cycle it.

The upload procedure uses UART, and it requires the external clock to be stable and accurate for uploading to work. If you have a signal generator, could you try to apply 16 MHz clock pulse straight into the XTAL1 pin on a chip where the bootloader seems "defective"? To be honest, I have no idea why you're experiencing this issue. I know there are many users using MiniCore and a bootloader on their 328PB boards, but none have reported issues similar to this one before.

Tsamsiy, 2021-04-01:
We do have a signal generator, but i'm not sure if that one is capable of such high frequencies...
One of these integrated clock modules with metal casing should do it..?

I'll post my results here once i have them.  Could take some time though... kinda busy at the moment.

> Since the device didn't have a reset button, I had to power cycle it.

I can remember that neither resetting nor power cycle did anything. The issue only disapperared afer reflashing the bootloader.

ameeli33, 2021-10-08:
HI
i have another problem whih burning bootloader by arduini as isp ..so i do it by usbasp programmer

closed 2022-08-19 by MCUdude