MCUdude/MiniCore #185: i cant use irremote library in the MiniCore

satheeshk787, 2021-04-01:
In file included from C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRremote.h:183:0,
                 from C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\examples\ReceiveDemo\ReceiveDemo.ino:60:
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h: In member function 'void IRrecv::enableIRIn()':
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:152:5: error: 'timerConfigForReceive' was not declared in this scope
     timerConfigForReceive();
     ^~~~~~~~~~~~~~~~~~~~~
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:153:5: error: 'TIMER_ENABLE_RECEIVE_INTR' was not declared in this scope
     TIMER_ENABLE_RECEIVE_INTR;  // Timer interrupt enable
     ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:154:5: error: 'TIMER_RESET_INTR_PENDING' was not declared in this scope
     TIMER_RESET_INTR_PENDING;   // NOP for most platforms
     ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h: In member function 'void IRrecv::disableIRIn()':
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:168:5: error: 'TIMER_DISABLE_RECEIVE_INTR' was not declared in this scope
     TIMER_DISABLE_RECEIVE_INTR;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\SATHEESH\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.1\cores\MCUdude_corefiles/Arduino.h:30:0,
                 from C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\examples\ReceiveDemo\ReceiveDemo.ino:33:
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h: At global scope:
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:1248:1: error: expected unqualified-id before 'void'
 ISR () // for functions definitions which are called by separate (board specific) ISR
 ^
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:1248:1: error: expected ')' before 'void'
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:1248:1: error: expected unqualified-id before 'void'
 ISR () // for functions definitions which are called by separate (board specific) ISR
 ^
C:\Users\SATHEESH\Documents\Arduino\libraries\IRremote\src/IRReceive.cpp.h:1248:1: error: expected ')' before 'void'
exit status 1
Error compiling for board ATmega328.

MCUdude, 2021-04-01:
Can you provide a link to the library and tell me what example you're trying to compile?

satheeshk787, 2021-04-19:
https://www.arduinolibraries.info/libraries/i-rremote

MCUdude, 2021-04-19:
I'm able to compile the SendDemo.ino sketch for ATmega328P. You have to enable LTO in the tools menu

satheeshk787, 2021-04-19:
> I'm able to compile the SendDemo.ino sketch for ATmega328P. You have to enable LTO in the tools menu

Ok.. I will try.

tomaskovacik, 2021-04-19:
it works, I reported in #179

satheeshk787, 2021-04-20:
Now It's working great. Thanks!!!!

closed 2021-04-20 by satheeshk787

tomaskovacik, 2021-04-20:
I think that LTO should be enabled by default as is enabled in Arduino official core ....