MCUdude/MiniCore #186: How to access PORTE in Arduino IDE?

geokscott, 2021-04-01:
This might not be the right place to ask this, but here goes.

Using the Arduino IDE, how would I access Port E pins?

Something like: digitalWrite(PIN_PE0, HIGH);
for pin 3 on the chip???

Thanks for any help!

MCUdude, 2021-04-01:
I'll assume you're working with a *PB chip, an ATmega328PB for instance?

There are several ways you can "access" a pin. One way is to use digitalWrite and pin PIN* macros, just as you've shown:

``` c++
digitalWrite(PIN_PE0, HIGH); // Set PE0 (Arduino pin 23) high
digitalWrite(23, HIGH);      // Set Arduino pin 23 (PE0) high
PORTE |= (1 << PE0);         // Set PE0 high, ignore the rest of the pins on the port

Note that Arduino pin numbers (like you have on an Arduino UNO) and the physical pin numbers on the chip are not the same and doesn't line up.

geokscott, 2021-04-01:
Yes, that's exactly what I'm looking for!  So where can I find a reference of Arduino IDE "pins" to the chip pins?  Nothing about these chips on the Arduino website.  Where do I find that 23 equates to PE0 and pin 3 on the PB chip?

Thanks!

MCUdude, 2021-04-01:
You can find all the relevant pinout in the MiniCore readme:

https://github.com/MCUdude/MiniCore#pinout

geokscott, 2021-04-01:
Ah!  Thanks you, it was right in my face and I didn't see it!  Just didn't look hard enough!  Thanks again!!!

MCUdude, 2021-04-01:
No problem! I might have put them a little too far down; so much to document...

> Thanks again!!!

You're welcome, thanks for using MiniCore!

closed 2021-04-01 by MCUdude