MCUdude/MiniCore #187: 2nd I2C on ATMega328PB handled different to Arduino IDE defaults !?

Balze2, 2021-04-19:
Hi MCUdude,

for some time I try to get the second I2C bus from the ATMega328PB working with display libraries (I.e. [U8G2](https://github.com/olikraus/u8g2))
I only get it working with a workaround.
It seems, that it's not working, because your core handles the second I2C different from the Arduino defaults.

A discussion with the developer of U8G2 showed, that his library needs a define of WIRE_INTERFACES_COUNT >= 2 to be able to work with another than the first I2C.
This value does not exist in your core (At least I wasn't able to find it. :)

The "Arduino Zero" is one of the few Microcontrollers with more than one I2C and official support. (Imho)

In the [variants.h](https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/variant.h) there is a define for the number of I2Cs.
The generation of Wire instances is different to your solution too. Please have a look at [wire.h](https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/Wire/Wire.h) and [wire.cpp](https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/Wire/Wire.cpp)

wire.h
`#if WIRE_INTERFACES_COUNT > 0
  extern arduino::TwoWire Wire;
#endif
#if WIRE_INTERFACES_COUNT > 1
  extern arduino::TwoWire Wire1;
#endif
#if WIRE_INTERFACES_COUNT > 2
  extern arduino::TwoWire Wire2;
...
...`

wire.cpp
`#if WIRE_INTERFACES_COUNT > 1
  arduino::TwoWire Wire1(&PERIPH_WIRE1, PIN_WIRE1_SDA, PIN_WIRE1_SCL);
  void WIRE1_IT_HANDLER(void) {
    Wire1.onService();
  }
#endif
#if WIRE_INTERFACES_COUNT > 2
  arduino::TwoWire Wire2(&PERIPH_WIRE2, PIN_WIRE2_SDA, PIN_WIRE2_SCL);
  void WIRE2_IT_HANDLER(void) {
    Wire2.onService();
  }
#endif
...
...`

Maybe it's an opinion to adapt this "Arduino - standard"?!

Thank you very much for your effort.

Best regards

Rainer

MCUdude, 2021-04-19:
Hi!

Are you able to use the second i2c port (`Wire1`) on ATmega328PB with u8g2 if you add `#define WIRE_INTERFACES_COUNT 2` to [pina_arduino.h](https://github.com/MCUdude/MiniCore/blob/master/avr/variants/pb-variant/pins_arduino.h)?

Balze2, 2021-04-19:
Hi,

good idea. Didn't try it yet. 🤣 
I'll try in the evening, when I'm back home.

Balze2, 2021-04-19:
Thank you very much for your hint!

That simple... shame on me 😄 

I added
`#define WIRE_INTERFACES_COUNT 2`
to the pins_arduino.h in the variants folder of the MiniCore.
And added
`#include <wire1.h>`
to theU8G2 "hello world" Example

Now it's working!

Thanks and best regards

Balze

MCUdude, 2021-04-19:
Thanks for testing! I'll add `#define WIRE_INTERFACES_COUNT 2` to pins_arduino.h then.

Thanks for reporting!

Balze2, 2021-04-19:
Thank you for taking care!

It's not necessary to include wire1.h!
This is obviously done by dependecies.

commit bd19d05f (2021-04-19): Add WIRE_INTERFACES_COUNT macro
Apparently, this is used to determine if a target has more than one i2c port, which the 328PB absolutely has. Closes #187
closed 2021-04-19 by MCUdude